stella-cv/stella_vslam #3: Merge bug fixes

ymd-stella, 2021-01-30:
This is a meta-ticket.
Some PRs have not been reviewed.

- [ ] https://github.com/xdspacelab/openvslam/pull/306
- [ ] https://github.com/xdspacelab/openvslam/pull/339
- [ ] https://github.com/xdspacelab/openvslam/pull/345
- [ ] https://github.com/xdspacelab/openvslam/pull/393
- [ ] https://github.com/xdspacelab/openvslam/pull/408
- [ ] https://github.com/xdspacelab/openvslam/pull/484
- [ ] https://github.com/xdspacelab/openvslam/pull/514
- [ ] https://github.com/xdspacelab/openvslam/pull/518
- [ ] https://github.com/xdspacelab/openvslam/pull/519

mirellameelo, 2021-03-14:
Hi! Is this done or some still missing?

ymd-stella, 2021-03-14:
If the item is not checked, it has not been merged.

jeffin07, 2021-08-03:
Hi,
I would like to work on this.I  am  currently exploring openvslam using docker can you please guide me to work on this issue

ymd-stella, 2021-08-03:
@jeffin07 I just checked, and there were only two left. I've added a link to the commit in the description, so please refer to it. The minimum you need to do is to apply these commits. It is also better if you can understand the cause of the problem and fix the areas where similar problems occur.

closed 2021-08-06 by ymd-stella