stella-cv/stella_vslam #6: Release with Bloom

ymd-stella, 2021-01-30:
ref. https://index.ros.org/doc/ros2/Tutorials/Releasing-a-ROS-2-package-with-bloom/

SteveMacenski, 2021-02-12:
I can help with this too if you have questions

ymd-stella, 2021-02-13:
My plan is as follows.

- Split the ros directory part of openvslam into openvslam_ros
- Separate the ROS version from the ROS2 version by branch
- Update the documentation for use with ROS/ROS2
- Update the Dockerfile
- Update the CI workflow
- Update openvslam for release on Bloom
  - Add package.xml
  - Follow the instructions in https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease
- Update openvslam_ros for release on Bloom
  - Follow the instructions in https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease

ymd-stella, 2021-02-13:
https://github.com/introlab/rtabmap is a good reference to do it.

SteveMacenski, 2021-02-13:
Makes sense - sounds like you've worked with bloom before. Not everyone has and I know it can be a _real_ pain to get up to speed on for your first few times. 

In case its helpful, I'll just leave this here. This is my tl;dr Bloom guide as requested by others in the past that gives a plain-english step-by-step guide without all the options listed up front. Sometimes its useful just to have a "here's best practices, if you do this, things will be streamlined" https://gist.github.com/SteveMacenski/1c321d1c9edca096ae4d763d8327c2ee

ymd-stella, 2021-02-21:
I' m considering how to resolve the dependency on the modified DBoW2.

ymd-stella, 2021-02-21:
It seems that in order to release it for Ubuntu 18.04 it needs to be modified to work with OpenCV 3.2.

ymd-stella, 2021-02-21:
I'm considering adding DBoW2 as a submodule like https://github.com/leggedrobotics/darknet_ros.

ymd-stella, 2021-02-21:
> It seems that in order to release it for Ubuntu 18.04 it needs to be modified to work with OpenCV 3.2.

In order to use the new versions of OpenCV and libg2o, I have decided to release openvslam for noetic/foxy.

SteveMacenski, 2021-02-23:
I'm not sure - anything openCV related if the wrong version is used turns into a complete nightmare

ymd-stella, 2021-02-24:
OpenVSLAM requires OpenCV 3.3.1 or later. (4 or later is also available.)
Ubuntu 18.04 does not meet this requirement as OpenCV 3.2 is used, but Ubuntu 20.04 meets this requirement as OpenCV 4.2 is used.

SteveMacenski, 2021-02-24:
Ah ok, and you want 18.04 to support Melodic? My uses (for Nav2) is all in 20.04 (since all ROS2 distributions I believe are on 20.04 now as well as Noetic)

ymd-stella, 2021-03-14:
@SteveMacenski May I ask you a question?
I have run bloom-release. The tags has been pushed to https://github.com/OpenVSLAM-Community/openvslam-release.

However, the automatic creation of the pull request failed.
`Failed to open pull request: KeyError - 'url'`

What is the pull request in the last step of bloom-release?
If I want to do this step manually, where and what kind of pull request should I create?

SteveMacenski, 2021-03-15:
I haven't seen that particular error before, it really would depend when that error appeared in the process to know for sure. The last step is that it forks `rosdistro` and adds the entry with all of the information (and a few more questions about PR testing and documentation statuses) to a branch submitted against `rosdistro` which will trigger the build farm binary building process on the release repo / versions (amongst other stuff)

ymd-stella, 2021-03-15:
https://github.com/ros/rosdistro/pull/28651

It worked. It seems that I tried to enter the document and source information halfway, which was a bad idea.

SteveMacenski, 2021-03-15:
Once it gets in, make sure to check in on https://build.ros2.org/ for the build status so you can make sure it built correctly / fix any issues. Usually on a first release to a new distro, I expect there to be an issue or two.

ymd-stella, 2021-03-15:
> Once it gets in, make sure to check in on https://build.ros2.org/ for the build status so you can make sure it built correctly / fix any issues.

I'm at a loss as to where to find related information.

SteveMacenski, 2021-03-16:
Yes, I use search :-)

ymd-stella, 2021-03-20:
The build is currently failing. There are three reasons 
- [x] fbow library file is not found in the second build.
- [ ] Unable to build fbow with -DCMAKE_BUILD_TYPE=None
- [ ] Unable to build fbow with aarch64

Items that have been fixed are checked.

ymd-stella, 2021-03-20:
build.ros2.org builds with -DCMAKE_BUILD_TYPE=None, which causes performance problems. There are three approaches to solve this problem. Which is best?

- Add optimization options when `-DCMAKE_BUILD_TYPE=None` is specified.
- Specify `-DCMAKE_BUILD_TYPE=Release`.
- Overwrite CMAKE_BUILD_TYPE with Release if `-DCMAKE_BUILD_TYPE=None`.

ymd-stella, 2021-03-20:
I took the first approach.

The performance appears to be worse in the release build, and I' m investigating the cause.

ymd-stella, 2021-03-21:
> The performance appears to be worse in the release build, and I'm investigating the cause.

It was caused by `-ffast-math`.

SteveMacenski, 2021-03-22:
The build farm should be using Release builds when none is specifically specified or overrided

ymd-stella, 2021-03-23:
There are four different build logs, but I don't know what each one does.
- Fdev__openvslam__ubuntu_focal_amd64
- Fsrc_uF__openvslam__ubuntu_focal__source
- Fbin_uF64__openvslam__ubuntu_focal_amd64__binary
- Fbin_ubv8_uFv8__openvslam__ubuntu_focal_arm64__binary

The following log was included, so it seems that the build type is set to `None` in `Fbin*`.

```
cd obj-aarch64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_AUTOGEN_VERBOSE=ON -DCMAKE_INSTALL_LIBDIR=lib/aarch64-linux-gnu -DCMAKE_INSTALL_PREFIX=/opt/ros/foxy -DCMAKE_PREFIX_PATH=/opt/ros/foxy ..
```

SteveMacenski, 2021-03-23:
dev is the master branch of your repo, source is the source build of the code for packaging, and the bin builds are the binary builds for deployment to `apt`