stella-cv/stella_vslam #18: Build scripts and vcpkg

Cambloid, 2021-01-31:
The original author created some powershell scripts to build openvslam on windows

https://github.com/shinsumicco/openvslam-on-msvc

I tried with vcpkg but it fails when setting the -DDBoW2_DIR in the command line.
It works when DBoW2_DIR path is set in the cmake script.

Is this caused by vcpkg?

ymd-stella, 2021-01-31:
> I tried with vcpkg but it fails when setting the -DDBoW2_DIR in the command line.
It works when DBoW2_DIR path is set in the cmake script.

I would like to see details of the procedure that fails and the procedure that succeeds.
I may be able to give you some advice, or someone else may be able to.

> Is this caused by vcpkg?

I don't know, I've never tried to build on Windows.
I'm not even sure how useful it would be to use vcpkg.

Contributions are welcome.
But we can't safely merge them until the Windows version of CI is ready.
I'm considering using CircleCI.

Cambloid, 2021-02-03:
vcpkg makes it much easier especially when you have a cmake project and many dependencies
the issue with DBoW2_DIR got resolved with cmake presets

I just made a pull request wih a possible solution:
https://github.com/OpenVSLAM-Community/openvslam/pull/23

ymd-stella, 2021-02-03:
@Cambloid It seems that the change `find_package(SuiteSparse CONFIG)` is causing the error `cs.h: No such file or directory` .

ymd-stella, 2021-02-03:
I reverted the changes in CMakeLists.txt.
Please let me know if you have a successful build using vcpkg without affecting existing builds.

closed 2021-02-03 by ymd-stella

Cambloid, 2021-02-04:
the build fails with the missing include file.

would something like this be acceptable?:
    if(MSVC)
	find_package(SuiteSparse CONFIG)
    else()
        find_package(SuiteSparse)
    endif()