stella-cv/stella_vslam #19: Setting up CI for windows and mac

ymd-stella, 2021-02-01:
It is related to #7 and #18.
I'm considering CircleCI.

ThomasHauth, 2021-04-16:
Hi @ymd-stella,
can I help to set this up?
Cheers,
Thomas

ymd-stella, 2021-04-16:
Thank you for your help.
I think https://github.com/OpenVSLAM-Community/openvslam/blob/main/.travis.yml is a good reference for CI for macOS.

- https://circleci.com/docs/2.0/hello-world-macos/
- https://circleci.com/docs/2.0/hello-world-windows/

ThomasHauth, 2021-04-16:
Is there a reason to not use TravisCI any more ?

ymd-stella, 2021-04-16:
There was a problem with the build taking a very long time. If you can solve this, we can use TravisCI.

https://travis-ci.org/github/OpenVSLAM-Community/openvslam/builds/756913945

CircleCI or GitHub Actions may be a good alternative.

ymd-stella, 2021-04-16:
After additional research, I found that we may need to support newer versions of Mac OS.

ymd-stella, 2021-04-17:
I've tried GitHub Actions and it seems to be working well.
It took 7 minutes to complete the installation of dependent packages without cache.
@ThomasHauth If you can help, please use GitHub Actions.