stella-cv/stella_vslam #26: Documentation of the community version

ymd-stella, 2021-02-05:
- [ ] Update README.md
- [ ] Update docs/

ref. https://readthedocs.org/

ymd-stella, 2021-02-06:
I made it.
openvslam-commnunity.rtfd.io

SteveMacenski, 2021-02-12:
@ymd-stella there's an extra `n` in your URL

Also, can you highlight why this community version? Is the original version's maintainers non-responsive and weren't willing to add others as maintainers?

We're looking at using OpenVSLAM as the formal integration for VSLAM for navigation2 in the next couple of months

ymd-stella, 2021-02-13:
> Is the original version's maintainers non-responsive and weren't willing to add others as maintainers? 

I have asked the community a question and until now there has been no response from the maintainers of the original version.
https://spectrum.chat/openvslam/general/what-do-you-think-about-a-community-fork~be231238-c7d5-478c-b3dd-61d437d101ad

SteveMacenski, 2021-02-13:
Makes sense, maybe worth emailing them from the emails in the package.xml? I just want to make sure that reasonable attempts to contact the developers are made since this would be a long-term commitment to maintain this and it would be a pain if the original developers came back after some new developments and this was out of sync. I would think you and they would both prefer to work in the main organization / repository.

ymd-stella, 2021-02-14:
I have sent them an email.
As the main maintainers appear to have been students, it is possible that their email addresses have become invalid after they have graduated.
In fact, for some email addresses, an error was returned.

SteveMacenski, 2021-02-14:
Ooof, that doesn’t bode well. I’m just shocked that they’d build such a fantastic, well designed, and highly performant VSLAM like this and just walk away. 

Well, I think if you don’t hear back from the couple of email addresses that didn’t bounce back, then I think you did all you reasonably could.

SteveMacenski, 2021-03-05:
@ymd-stella did #40 complete part of this?

SteveMacenski, 2021-03-10:
@ymd-stella what specifically in `/docs` needs to be updated? I could work on that tomorrow afternoon.

SteveMacenski, 2021-03-10:
I think that closes out this ticket. Grepping for xdspace only gives the notification that this is a community fork

closed 2021-03-11 by ymd-stella