stella-cv/stella_vslam #38: Support Stereo and RGBD in ROS2

SteveMacenski, 2021-02-26:
Hi,

I notice only monocular is supported in ROS2 (with ROS1 having all 3). This is something me and my colleague can help commit to implementing as part of the Nav2 / ROS2 integration work. We'll likely have other features / things we want to expose (like adding a feature map publishing topic, among others) but figured this is a good place to get the ball rolling.

mirellameelo, 2021-03-02:
@SteveMacenski, there are new files modification regarding the stereo support for ROS2 in my last [PR](https://github.com/OpenVSLAM-Community/openvslam/pull/39). I appreciate it if you could review it. I used the @ymd-stella work as a base and follow the same approach. 
I think it would be a good idea to tag the next steps with respect to ROS2.
Thanks in advance.

SteveMacenski, 2021-03-02:
@mirellameelo I think you should split your PR into 2. Pushing too much in a PR makes things slow down dramatically. One for doc updates and other for a big feature update would be best so that @ymd-stella can quickly review

SteveMacenski, 2021-03-02:
A concern I have here is that the ROS wrapper isn't making use of ROS time, so there's going to be issues in synchronizing the outputs of OpenVSLAM with the rest of a ROS-based system since the timing of the output poses and transforms may not be from the same clock

mirellameelo, 2021-03-02:
Thanks for replying. I agree regarding the timers and loggings. I'll change it. 
Alghtout there are ROS2 packages responsible to publish frames, I think it is interesting just from a didactic point of view to keep this "publisher". These ones are aligned with the example to run in ROS. So I believe is a practical and easy way to do at least a test, especially for new users, besides its use as a reference to add new publishers for different dataset patter.

SteveMacenski, 2021-03-04:
For those in the Nav2 slack group, we're discussing on the mapping and localization channel how to port ROS2 + expose all the stuff necessary.

mirellameelo, 2021-03-04:
@SteveMacenski please take a look at [this comment](https://discourse.ros.org/t/new-navigation-slack/13845/9). I'm facing the same. Tks in advance

SteveMacenski, 2021-03-04:
Also working on making the parameters ROS-y so that people can modify the OpenVSLAM settings like any other ROS node

mirellameelo, 2021-03-04:
@SteveMacenski I'm facing [this](https://discourse.ros.org/t/new-navigation-slack/13845/9?u=mirellameelo). I just sent you a PM via LinkedIn.

AlexeyMerzlyakov, 2021-03-05:
Copying the task plan from Slack discussion:

[In progress by @mirellameelo in #41] **task1**: Support for Stereo cameras in ROS2 OpenVSLAM
**task2**: Support for RGB-D cameras in ROS2 OpenVSLAM
[Done by @mirellameelo in a repository fork] **task3**: Also, it is need to add TF2 in there. Currently, there is nav_msgs::msg::Odometry pose messages are being published. I think we need to have supplementary full "map->odom" frames transform for ROS2 SLAM/localization nodes.

Also the rectification for calibrated cameras is made only for ROS1, so we have to:
**task4**: Add camera images rectification into ROS2 slam/localization nodes.
Here is a few open questions: How could we verify the camera calibrations? Can we use a Gazebo simulator, say on TurtleBot3 model?
Does this model support a non-calibrated cameras simulation?
Real H/W verification also does not hurt.
All these questions are in the scope under investigation of this task.

After all nodes will be prepared, this might be an extra activity:
**task5**: Make unification process of what we've done. This will be required to get one ROS2 node acting in both SLAM or Localization modes (selected by ROS2 parameter), and also supporting all types of previously elaborated cameras (mono/stero/rgbd).

Currently, OpenVSLAM has only internally used map database which is not ROS2 representative. Therefore we need
**task6**: Investigate and extract Occupancy Grid map (and/or PCL map reconstuction) from OpenVSLAM internals.
This could be divided into 2 parts (@SteveMacenski suggestion):
- A) expose the feature map to the user as a pointcloud (probably, XYZ, color, probably the point of origin, any other metadata associated with it) and
- B) how do use it to do useful things to compute rough occupancy information as a separate node in Nav2. So these are really 2 tasks interfacing over a topic

For this task I have a personal interest, since I previously was touched OccGrid maps making in a Visual SLAM and also having an experience of working with maps (map server re-designs and costmap filters development).

**task7**: Based on above task there should be prepared probably documentation update for OpenVSLAM

ymd-stella, 2021-03-08:
The ROS wrappers have been migrated to an independent repository https://github.com/OpenVSLAM-Community/openvslam_ros.
Please contribute there. If you want to create a new ROS issue, please create it in openvslam_ros.
The issues that already exist can be left.
The ros directory will be removed after confirming that there is no problem and updating the documentation.

SteveMacenski, 2021-03-08:
@ymd-stella should you remove the `ros/` directory in `openvslam` then? 

I see a `main` branch, what version is that? I think you can delete `ros` since `noetic-devel` is the last ROS1 so there's nothing else to add to later. For now. I'd recommend also removing `foxy-devel` until we have something for ROS2 working on `ros2` worth releasing to foxy (e.g. its alot of effort to keep those in sync while we're still developing it)