stella-cv/stella_vslam #48: Baidu Wangpan link is not available

Airplane5, 2021-03-10:
Except Google Drive and Baidu Wangpan, how can I get the test dataset?

SteveMacenski, 2021-03-10:
Actually the vocab drive link is just totally dead - @ymd-stella do you have that locally we can reupload somewhere?

mirellameelo, 2021-03-10:
@Airplane5 do you mean these datasets?
[Equirectangular images](https://drive.google.com/drive/folders/1A_gq8LYuENePhNHsuscLZQPhbJJwzAq4)
[Fisheye](https://drive.google.com/drive/folders/1SVDsgz-ydm1pAbrdmhRQTmWhJnUl_xr8)
[All](https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html)

Airplane5, 2021-03-10:
@mirellameelo 
Thank u for reply!
I can't connect google drive because it's banned in China.

mirellameelo, 2021-03-10:
@Airplane5 well, if there is any other way I can send you at least one of these videos while it is not solved, let me know. =/

Airplane5, 2021-03-10:
@mirellameelo  So nice of u. thank u again.

ymd-stella, 2021-03-10:
> do you have that locally we can reupload somewhere?

It exists in the docker image for CI.

SteveMacenski, 2021-03-10:
Can you point to where? I just looked at all the docker files in this repo and I don't see a reference to these files. I think I see something in the travis file, but that's not executed

SteveMacenski, 2021-03-10:
This is a good question. A GCP or AWS bucket seems like the most reasonable place I think people can access it from via `wget` in the US and China, though that's not a free option. Drive is a reasonable free option, but has issues in China. 

ORB-SLAM directly inputs it as a tarball in their git repo https://github.com/raulmur/ORB_SLAM2/blob/master/Vocabulary/ORBvoc.txt.tar.gz which technically works too, I think github has a limit of 100MB per repo, so that keeps it under.

Maybe my recommendation would be adding a new repo in the community org that literally just hosts this file so that its not in the main one (to help cloning times). We might eventually have an issue if too many people are downloading it with bandwidth limitations, but one issue at a time. Git does have large file storage (LFS) options, but they're quite expensive tiered by bandwidth we can't control directly.

SteveMacenski, 2021-03-12:
https://github.com/OpenVSLAM-Community/DBoW2_orb_vocab/blob/main/orb_vocab.dbow2 file is now there!

ymd-stella, 2021-03-12:
I don't think we can close it yet.
What @Airplane5 is referring to is the dataset, not the vocabulary file.

Airplane5, 2021-03-12:
@SteveMacenski 
thank u!
I get the BoW right now.
But dataset is still out of reaching.
Could you give a link that I can access in China mainland if convenient?

SteveMacenski, 2021-03-12:
I dont think anyone on this thread can (or someone could). The data is on google drive and we dont have better options for files of that size. Try a VPN?

Airplane5, 2021-03-14:
I will try using VPN. You can close now.
Thank you again!

closed 2021-03-14 by ymd-stella