stella-cv/stella_vslam #49: Remove link to spectrum chat?

SteveMacenski, 2021-03-10:
Hi,

I glanced through that spectrum and no one's answered anything as far as I could scroll. Maybe we should just put that to rest and remove links to it now that the official maintainers have given up on the project entirely.

ymd-stella, 2021-03-10:
I don't think it's not working. Are you sure you're not misunderstanding something?

ymd-stella, 2021-03-10:
I watched as xdspacelab/openvslam was flooded with trivial questions. I do not agree with the removal of the community.

closed 2021-03-10 by ymd-stella

SteveMacenski, 2021-03-10:
@ymd-stella I didn't say it wasn't working, I'm saying its not useful. If no one is answering questions, its just people screaming into the void with problems thinking someone will help (that isn't coming).

ymd-stella, 2021-03-10:
Even if no one answers, it helps in the following two ways

- We can remove trivial questions from github issues
- We can learn about what users are likely to get wrong and what they want to achieve

In addition, there is actually a useful exchange of information.

- https://spectrum.chat/openvslam/general/gets-stuck-in-a-single-position-if-left-still~1feef43c-6fae-4687-8fd9-89e9dccc9f3a
- https://spectrum.chat/openvslam/beginner/camera-is-facing-the-wrong-direction~0ea1a2a6-4d8b-4704-807d-40fce56999b7
- https://spectrum.chat/openvslam/beginner/slam-with-ros-results-are-poor-but-works-fine-on-standard-installation~42725467-3590-4d69-ace7-3baa40923366

The essence of the problem is that there are not enough users in the community who can provide answers.
It is certain that it cannot be solved by closing the spectrum community.

surfii3z, 2021-03-11:
@ymd-stella  How about discussion in github? there is a new feature for that?