stella-cv/stella_vslam #63: Timeline for IMU

oscarpang, 2021-03-30:
Thanks to all who contributed to this project. Also thanks for the community fork.

Is there a timeline for integrating the IMU functionality? Or a WIP branch? There are a number of cameras now equipped with the IMU(intel RealSense, StereoLab's Zed). 

Much thanks
Weiwu

ymd-stella, 2021-04-03:
https://github.com/OpenVSLAM-Community/openvslam/tree/feat/imu

ymd-stella, 2021-04-03:
My working branch is https://github.com/ymd-stella/openvslam-community/tree/wip/imu.
Contributions and comments are welcome.

ymd-stella, 2021-04-03:
@oscarpang Did you get the information you needed?

oscarpang, 2021-04-05:
Yes. Thanks so much!

closed 2021-04-05 by oscarpang

oscarpang, 2021-04-06:
@ymd-stella do you have a pointer to how to specify the transform from IMU to camera? Also, to specify the frequency of the IMU?

oscarpang, 2021-04-06:
I actually found the example. Thanks again.