stella-cv/stella_vslam #68: Documentation

surfii3z, 2021-04-15:
Hi, thanks for maintaining the openVSLAM.

I try to fresh install using the OpenVSLAM-Community documentation on [Installation section](Installation), however, I found that it does not build the executable like `run_video_slam`, etc. This makes it hard to simply follow the [tutorial](https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html) here

ymd-stella, 2021-04-15:
You need to run `sudo make install` after `make -j4`.

ymd-stella, 2021-04-15:
The corresponding file is located at https://github.com/OpenVSLAM-Community/openvslam/blob/main/docs/installation.rst.

ymd-stella, 2021-04-15:
@surfii3z Do you plan to fix it?

surfii3z, 2021-04-15:
> @surfii3z Do you plan to fix it?

 @ymd-stella Thanks for your reply.

Yeah, sure I could do that. 

But actually I kind of don't like the idea of installing openvslam lib though.

Couldn't we make it self-contain?

surfii3z, 2021-04-15:
> You need to run `sudo make install` after `make -j4`.

For the note, after doing this, it still has the this error 

```bash
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Pangolin_Viewer_LIB
    linked by target "run_slam" in directory /home/jy/test_ws/src/openvslam_ros/src
    linked by target "run_localization" in directory /home/jy/test_ws/src/openvslam_ros/src

-- Configuring incomplete, errors occurred!
See also "/home/jy/test_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jy/test_ws/build/CMakeFiles/CMakeError.log".
```

ymd-stella, 2021-04-16:
`-DINSTALL_PANGOLIN_VIEWER=ON` is required.
https://github.com/OpenVSLAM-Community/openvslam/blob/7392f6f144e52e829ba22167dce119d3b7b01466/src/pangolin_viewer/CMakeLists.txt#L31

ymd-stella, 2021-04-16:
Build openvslam with the following command.

```
cmake \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DINSTALL_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DBOW_FRAMEWORK=DBoW2 \
    -DBUILD_TESTS=ON \
    -DBUILD_EXAMPLES=ON \
    ..
```

surfii3z, 2021-04-16:
> Install openvslam with the following command.
> 
> ```
> cmake \
>     -DUSE_PANGOLIN_VIEWER=ON \
>     -DINSTALL_PANGOLIN_VIEWER=ON \
>     -DUSE_SOCKET_PUBLISHER=OFF \
>     -DUSE_STACK_TRACE_LOGGER=ON \
>     -DBOW_FRAMEWORK=DBoW2 \
>     -DBUILD_TESTS=ON \
>     -DBUILD_EXAMPLES=ON \
>     ..
> make -j4
> sudo make install
> ```

@ymd-stella  thanks for your reply.

```bash
Base path: /home/jy/test_openvslam_ws
Source space: /home/jy/test_openvslam_ws/src
Build space: /home/jy/test_openvslam_ws/build
Devel space: /home/jy/test_openvslam_ws/devel
Install space: /home/jy/test_openvslam_ws/install
####
#### Running command: "cmake /home/jy/test_openvslam_ws/src -DUSE_PANGOLIN_VIEWER=ON -DUSE_SOCKET_PUBLISHER=OFF -DCATKIN_DEVEL_PREFIX=/home/jy/test_openvslam_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jy/test_openvslam_ws/install -G Unix Makefiles" in "/home/jy/test_openvslam_ws/build"
####
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/jy/test_openvslam_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jy/test_openvslam_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - cv_bridge
-- ~~  - openvslam_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(cv_bridge)
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.17") 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
[... 94 lines cut ...]
[ 83%] Linking CXX shared library /home/jy/test_openvslam_ws/devel/lib/python2.7/dist-packages/cv_bridge/boost/cv_bridge_boost.so
[ 83%] Built target cv_bridge_boost
[ 91%] Linking CXX executable /home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_localization
[100%] Linking CXX executable /home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_slam
/usr/bin/ld: CMakeFiles/run_localization.dir/openvslam_ros.cc.o: undefined reference to symbol '_ZN9openvslam6systemC1ERKSt10shared_ptrINS_6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
//usr/local/lib/libopenvslam.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
openvslam_ros/src/CMakeFiles/run_localization.dir/build.make:181: recipe for target '/home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_localization' failed
make[2]: *** [/home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_localization] Error 1
CMakeFiles/Makefile2:2069: recipe for target 'openvslam_ros/src/CMakeFiles/run_localization.dir/all' failed
make[1]: *** [openvslam_ros/src/CMakeFiles/run_localization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/run_slam.dir/openvslam_ros.cc.o: undefined reference to symbol '_ZN9openvslam6systemC1ERKSt10shared_ptrINS_6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
//usr/local/lib/libopenvslam.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
openvslam_ros/src/CMakeFiles/run_slam.dir/build.make:181: recipe for target '/home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_slam' failed
make[2]: *** [/home/jy/test_openvslam_ws/devel/lib/openvslam_ros/run_slam] Error 1
CMakeFiles/Makefile2:2007: recipe for target 'openvslam_ros/src/CMakeFiles/run_slam.dir/all' failed
make[1]: *** [openvslam_ros/src/CMakeFiles/run_slam.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
```

It solved the Pangolin viewver issue. But it came to this error instead.

ymd-stella, 2021-04-16:
I've fixed it.
Please update openvslam_ros.

surfii3z, 2021-04-17:
> I've fixed it.
> Please update openvslam_ros.

Thanks @ymd-stella. Editting the CMakeList.txt fixed the issue.

closed 2021-04-17 by surfii3z

Gilaine, 2021-07-07:
Hello @ymd-stella,

After I made catkin_make and it finished successfully, I tried the following command but I gives me that error :

rosrun openvslam_ros run_localization     -v /path/to/orb_vocab.fbow     -c /path/to/config.yaml     --map-db /path/to/map.msg

/devel/lib/openvslam_ros/run_localization: error while loading shared libraries: libfbow.so.0.0: cannot open shared object file: No such file or directory

how can I fix this ?

ymd-stella, 2021-07-07:
@Gilaine Please ask in [Discussions](https://github.com/OpenVSLAM-Community/openvslam/discussions).

Additionally, there is not enough information for me to answer. At least about the OS and the build procedure you performed is needed.

Gilaine, 2021-07-07:
Thank you so much @ymd-stella