stella-cv/stella_vslam #76: Failed to load ORB Vocabulary in Docker

KelvinCPChiu, 2021-04-28:
I have successfully compiled the Openvslam in Ubuner 18.04 docker with the instruction in the link, 
https://openvslam-community.readthedocs.io/en/latest/docker.html#building-docker-images

After compilation, the following error message was given when I was trying to run the samples in the following link, 
https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html#chapter-simple-tutorial.

I have confirmed that I have downloaded the dbow2 file to the directory and change the -v accordingly. Still, the following error message occurred.

Thank you very much for your help !

[2021-04-28 09:11:24.241] [I] loading ORB vocabulary: ./orb_vocab.dbow2
terminate called after throwing an instance of 'std::runtime_error'
  what():  Vocabulary::fromStream invalid signature
*** Aborted at 1619601084 (unix time) try "date -d @1619601084" if you are using GNU date ***
PC: @     0x7f6fd8ab5fb7 gsignal
*** SIGABRT (@0x237) received by PID 567 (TID 0x7f6fdc984c00) from PID 567; stack trace: ***
    @     0x7f6fdbce1980 (unknown)
    @     0x7f6fd8ab5fb7 gsignal
    @     0x7f6fd8ab7921 abort
    @     0x7f6fd9336957 (unknown)
    @     0x7f6fd933cae6 (unknown)
    @     0x7f6fd933cb21 std::terminate()
    @     0x7f6fd933cd54 __cxa_throw
    @     0x7f6fd79e85ea fbow::Vocabulary::fromStream()
    @     0x7f6fd79e8c12 fbow::Vocabulary::readFromFile()
    @     0x7f6fdbf90f8a openvslam::system::system()
    @     0x5578356f5def (unknown)
    @     0x5578356f4b15 (unknown)
    @     0x7f6fd8a98bf7 __libc_start_main
    @     0x5578356f4fba (unknown)
Aborted (core dumped)

ymd-stella, 2021-04-28:
See https://github.com/OpenVSLAM-Community/openvslam/discussions/75. Documentation will be updated soon.

KelvinCPChiu, 2021-04-28:
> See #75. Documentation will be updated soon.

Thanks @ymd-stella. I was confused how to convert the FBow from DBoW2 for a while even I found that conversion code was provided in FBoW.

KelvinCPChiu, 2021-04-28:
> See #75. Documentation will be updated soon.

Thanks @ymd-stella. I was confused how to convert the FBoW from DBoW2 for a while even I found that conversion code was provided in FBoW.

closed 2021-04-28 by KelvinCPChiu