stella-cv/stella_vslam #106: free() invalid pointer

manosmaroulis, 2021-05-27:
Hello i tried openvslam with RICOH theta V, but for some reason i have a memory problem. I use rpi4 b 2GB RAM, 
Below i show the configuration on the terminal, thank you
[image]

ymd-stella, 2021-05-27:
You say you have not changed the code, but there are multiple examples. Which one is it?

We may get some hints by building openvslam with `-DUSE_STACK_TRACE_LOGGER=ON`.

It is recommended to experiment with open datasets at first. openvslam has not been tested with rpi4b.
ref. https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html#sample-datasets
The basic method in science and engineering is to unify the conditions with other experiments and change only one factor. This is the classic one-factor-at-a-time method. Unless you can find the minimum requirements to reproduce the problem, we cannot resolve this issue.

manosmaroulis, 2021-05-27:
Thanks for the reply, i ran the run_camera_slam with monocular camera RICOH THETA V, i will try the open datasets and give more feedback. Thank you again.

ymd-stella, 2021-12-24:
@manosmaroulis Any progress?

closed 2022-04-11 by ymd-stella