stella-cv/stella_vslam #108: Limitations of relocalization

ymd-stella, 2021-05-29:
The conditions for successful relocalization are difficult for non-specialists to understand. In this issue, we will consider adding documentation, logs, and visualization.
Close camera position and camera angle are relatively easy conditions to understand. This is essentially the presence of enough landmarks that can be matched with the features of the frame. In Equirectangular, the relocalization succeeds regardless of the yaw angle of the camera.

jeffin07, 2021-08-06:
@ymd-stella Hi can i work on this issue?currently I'm exploring slam so contributing will  help me to understand.I see documentation and other task which will be good for a beginner that's why i asked

ymd-stella, 2021-08-06:
We have divided the task into multiple issues. If you have any questions, please ask them in each issue.

If you have never run relocalization before, I suggest you try it. For example, if you run localization on a map created by running slam, relocalization will be performed during initialization. See https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html.