stella-cv/stella_vslam #142: Describes the rough flow of the relocalization process

ymd-stella, 2021-08-06:


jeffin07, 2021-08-15:
@ymd-stella  does  the rough workflow documentation follows the format of the docs folder (.rst format) ?

ymd-stella, 2021-08-16:
> Can you please tell me What do you mean the "rough flow" of relocalization process.

For example, it is something like the following:

https://github.com/OpenVSLAM-Community/openvslam/blob/c5818b5fdeecf7cb4ae11d0589660862d5c570ef/src/openvslam/module/relocalizer.cc#L63

https://github.com/OpenVSLAM-Community/openvslam/blob/c5818b5fdeecf7cb4ae11d0589660862d5c570ef/src/openvslam/module/relocalizer.cc#L73

https://github.com/OpenVSLAM-Community/openvslam/blob/c5818b5fdeecf7cb4ae11d0589660862d5c570ef/src/openvslam/module/relocalizer.cc#L90

ymd-stella, 2021-08-16:
> does the rough workflow documentation follows the format of the docs folder (.rst format) ?

Yes.

jeffin07, 2021-08-30:
@ymd-stella  Can we close this issue or is there any more additions  ?

ymd-stella, 2021-08-30:
No.

closed 2021-08-30 by ymd-stella