stella-cv/stella_vslam #159: Failed to find CXSparse

MitchAllain, 2021-09-17:
OS: Ubuntu 18.04
branch: `main` (6ce268c)

Following the installation instructions in the docs, after installing all deps, and running `cmake ..` for openvslam, I saw the error

```
Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
```

It appears that the debian for suitesparse installed by `sudo apt install libsuitesparse-dev` installs the `cs.h` header in `/usr/include/suitesparse/cs.h` rather than one of the `CXSPARSE_CHECK_INCLUDE_DIRS` listed in `FindCXSparse.cmake`

https://github.com/OpenVSLAM-Community/openvslam/blob/6ce268c29539eba8ba624abc39fd5d178c85d3d0/cmake/FindCXSparse.cmake#L127-L132

I modified the above file to include the search path `/usr/include/suitesparse` and the build succeeded. I'm not a CMake expert, so please confirm this an appropriate addition and I can issue a PR.

ymd-stella, 2021-09-17:
CXSparse is not SuiteSparse. Does your log look like the following?

```shell-session
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "5.1.2")
```

MitchAllain, 2021-09-17:
CXSparse is included in [suitesparse](https://github.com/DrTimothyAldenDavis/SuiteSparse), as is the missing header. Or is it meant to be installed separately? Below is my output.

```
mallain@laplace:build$ cmake     -DUSE_PANGOLIN_VIEWER=ON     -DINSTALL_PANGOLIN_VIEWER=ON     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRA
CE_LOGGER=ON     -DBUILD_TESTS=ON     -DBUILD_EXAMPLES=ON     ..
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization: DISABLED
-- Use OpenCV 3.4.0
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/local/lib/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/local/include
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is com
piled with timing).
-- Found METIS library: /usr/local/lib/libmetis.so.
-- OpenMP: DISABLED
-- SSE3 for ORB extraction: DISABLED
-- SSE for floating-point operation: DISABLED
-- BoW framework: FBoW (found in /usr/local/include)
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to prefe
rring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Stack trace logger: ENABLED
-- Google Perftools: DISABLED
-- Viewer for examples: PangolinViewer
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mallain/lmr_ws/src/vslam/openvslam/build/googletest-download
Scanning dependencies of target googletest-download
[100%] Built target googletest-download
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mallain/lmr_ws/src/vslam/openvslam/build
```

PR #161 (2021-09-19): Fix FindCXSparse.cmake
Close #159
closed 2021-09-19 by ymd-stella