stella-cv/stella_vslam #203: Stereo Rectifier for fisheye stereo camera

albertarla, 2021-11-23:
Hi everyone, I'm currently working with the camera Realsense T265C. Is a Stereo camera with fisheye lenses. When calling `openvslam::util::stereo_rectifier`  I've notice that it is only working for perspective lenses although in the code there's already the option to [rectify fisheye images](https://github.com/OpenVSLAM-Community/openvslam/blob/7ae27bea658d763b4846aebba18e3524feec5287/src/openvslam/util/stereo_rectifier.cc#L47) . Is there any reason for that?

albertarla, 2021-11-23:
At this [branch](https://github.com/albertarla/openvslam/tree/stereo_rectifier_fisheye) from my fork I've modified it to work with stereo fisheye lenses. It is needed to add these parameters to the config file:
```yaml
StereoRectifier:
  setup: stereo
  model: fisheye
  K_left: [286.922, 0.0, 419.159, 0.0, 287.006, 403.235, 0.0, 0.0, 1.0]
  D_left: [-0.00965215, 0.0463874, -0.0440295, 0.00834438]
  R_left: [1.0,0.0,0.0,
           0.0,1.0,0.0, 
           0.0,0.0,1.0]
```
If there's no reason to not add fisheye lenses to the stereo_rectifier I will create a PR of this branch. 

Thanks in advance

ymd-stella, 2021-11-23:
See https://github.com/OpenVSLAM-Community/openvslam/pull/164.

ymd-stella, 2021-11-23:
I've updated the documentation.
https://github.com/OpenVSLAM-Community/openvslam/commit/0ec0721a919176a3e43c702ded185881803911f5

closed 2021-11-23 by ymd-stella