stella-cv/stella_vslam #217: Add Multi-Camera support

slovak194, 2021-12-08:
Hi!

I'm currently investigating the possibilities to introduce multi-camera (fisheye) support. Are there any plans or thoughts regarding how to do that?
Here are some thoughts:
1. Multi-image keyframes, camera rig, and feature matching across all cameras inspired by [MultiCol-SLAM](https://github.com/urbste/MultiCol-SLAM)
2. Stereo matching without rectification step inspired by fisheye stereo mode from [ORB_SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3) or from [Real-Time Variational Fisheye Stereo without Rectification and Undistortion](https://arxiv.org/abs/1909.07545) in case of overlapping cameras.

I'd love to hear any suggestions regarding architectural and algorithmic work to be done.

Best regards,
Alex

dimaxano, 2021-12-08:
Hi!

I'm willing to help with that, but I am not familiar yet with multi camera slam approaches, need to study it deeply

Also, as one more source of inspiration, [that](http://rpg.ifi.uzh.ch/docs/ICRA20_Kuo.pdf) (PDF) ETH paper could be seen

slovak194, 2021-12-08:
Hi, @dimaxano!

> Also, as one more source of inspiration, that (PDF) ETH paper could be seen

Yes, I've seen this article before. It sounds really great. But the reference implementation is missing and it may be too complex to target automatic/adaptive detection of available cameras.

But it definitely worse looking into.

QMonteCarlo, 2022-05-11:
I'm currently working on this project as my personal project,porting from multicol-slam and openvslam.

ymd-stella, 2022-05-11:
> I'm currently working on this project as my personal project,porting from multicol-slam and openvslam.

MultiCol-SLAM is GPLv3. Therefore, if you ported their implementation, it cannot be merged into this repository. Please be careful.

Alex-Beh, 2022-09-12:
Hello, any update on this new feature? I am willing to help.

slovak194, 2022-09-12:
> Hello, any update on this new feature? I am willing to help.
I've seen @QMonteCarlo is working on something, but I have not found the for he is using.
It is still very relevant to me, but I have not progressed far.

Alex-Beh, 2022-09-13:
@QMonteCarlo @ymd-stella May I know your ideas on this feature? Any proposal or idea so far? else I can try to work on it and discuss with you all in future.

nnop, 2023-09-16:
For multiple camera, the problem would be MVS.