stella-cv/stella_vslam #223: Load second non-overlapping maps

ymd-stella, 2021-12-16:
See https://github.com/OpenVSLAM-Community/openvslam/discussions/188.

MartiBolet, 2021-12-16:
Hi @AlbertArla and I will try to implement it. The plan we made is:
1. Create a function in 'system' which can load a new map to an existing 'map_db'.
  1.1 For value "cameras" the camera of the new map will be added to the existing.
  1.2 When trying to append the new 'keyframes' and 'landmarks', using 'keyframe_next_id' and 'landmark_next_id' of the existing map, shift the ids of the new map to not collapse with the ids of the old map. Note the 'id' must be changed in the key of "landmarks" and "keyframes", but also in the references of each 'keyframe' to 'lms_ids' and for each 'landmark' the if of 'ref_keyfrm'
  1.3 Increase the 'keyframe_next_id' to the new one of the new map and 'landmark_next_id'
 1. Function defined in step 1 takes an argument of the offset to be applied between the two maps.
   2.1 Apply this transformation for each 'keyframe' and to each 'landmark' of the new map. Transform the field of 'rot_cw' and 'trans_cw' for keyframes, and 

What we don't have clear is how the two maps will interact, would the system localize in both maps or they have to be connected via graph?

ymd-stella, 2021-12-16:
> What we don't know is how the two maps will interact, would the system localize in both maps or they have to be connected via graph?

I don't expect to access information about keyframes and landmarks that are not connected by graphs from the reference keyframe, except during loop closing. If the relocalization is successful, the rest of the process will be almost the same as usual.