stella-cv/stella_vslam #232: Accelerated ORB feature extraction for Nvidia Jetson. ARM neon or CUDA?

slovak194, 2021-12-21:
Hi!

While the current implementation of the ORB extractor works very well on x86_64 many people use ARM for their robotics projects. Particularly, the well-known Nvidia Jetson platform is built on aarch64.

I see a few ways to improve performance on Jetson:
1. Use ARM neon instructions where possible instead of SSE/AVX
2. Use CUDA (works only for Jetson)

Both ways will require changes to the build system and CI pipeline to test.

Is there any preference of what is better to have first? Does anybody have a willingness to help?

Best regards,
Alex

bot-lin, 2023-02-13:
Hi @slovak194 I am wondering if you are still interested in implemeting the project on ARM?  I just read the work here (https://github.com/zanazakaryaie/ARM-VO). It will be interesting if we make it work for this project. I am not an expert on NEON but happy to discuss and learn more.

Regards
Zheng

Ashmit101, 2024-01-29:
Hello there,

I am trying to build this project using CUDA supported OpenCV so that I can use it on Jetson Nano. I do not understand this project completely but I tried migrating orb_extractor.cc file to use the functions of CUDA. It builds successfully but while running the program, I keep getting following error:

`error: (-217:Gpu API call) an illegal memory access was encountered in function 'computeOrbDescriptor_gpu'`

As long as it is running, I do not see any keypoint in the viewer.

Is there a simple explanation about the flow of the program? That could be very helpful.

sandeepdutta, 2024-05-14:
I am interested in this, did you manage to use CUDA accelerated OpenCV ?

Sandeep