stella-cv/stella_vslam #237: Process died, occured in commit f42bfcf

Sang-Beom-Woo, 2021-12-24:
Recently, I pulled the code with the latest version 
(openvslam commit code: f42bfcfabbaf3e4fd0021ac3e283bb89a4312db9
 openvslam_ros commit code:  a1e575aa4b811639a545644b438f5bf83cd17684 )

The updated codes are compiled successfully and installed in the dynamic link library
When I run the executable run_slam in ROS2, it works fine temporarily. However, run time errors occur at any random time without outputting any error messages.

Using gdb prefix, it shows me the following backtrace just before the process dies.

```
#0  0x00007ffff7b24b1d in openvslam::data::landmark::erase_observation(std::shared_ptr<openvslam::data::keyframe> const&) () at /usr/local/lib/libopenvslam.so
#1  0x00007ffff7bc437e in openvslam::optimize::local_bundle_adjuster::optimize(std::shared_ptr<openvslam::data::keyframe> const&, bool*) const () at /usr/local/lib/libopenvslam.so
#2  0x00007ffff7ae3dde in openvslam::mapping_module::mapping_with_new_keyframe() ()
    at /usr/local/lib/libopenvslam.so
#3  0x00007ffff7ae4904 in openvslam::mapping_module::run() () at /usr/local/lib/libopenvslam.so
#4  0x00007ffff6b62de4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7243609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff6850293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
I've experimented on [intel NUC core i7 10gen with 16GB RAM], [RYZEN 7 5800H with 32GB RAM] and they both show the same issue. Is there anyone who solved this problem?

ymd-stella, 2021-12-24:
What exceptions are being thrown?
In order to get information for debugging, I recommend to set `-DCMAKE_BUILD_TYPE=RelWithDebInfo` when executing cmake for building OpenVSLAM.

Sang-Beom-Woo, 2021-12-27:
The exceptions thrown by gdb was 
```
Thread 28 "run_slam" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb93b9700 (LWP 17507)]
__gnu_cxx::__atomic_add_dispatch (__val=1, __mem=0xbfde878c8ffe6133) at /usr/include/c++/9/ext/atomicity.h:96
96	      __atomic_add(__mem, __val);
```
additional information provided by gdb is shown as follows.
```
#0  __gnu_cxx::__atomic_add_dispatch (__val=1, __mem=0xbfde878c8ffe6133) at /usr/include/c++/9/ext/atomicity.h:96
#1  std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_weak_add_ref() (this=0xbfde878c8ffe6127) at /usr/include/c++/9/bits/shared_ptr_base.h:178
#2  std::__weak_count<(__gnu_cxx::_Lock_policy)2>::operator=(std::__weak_count<(__gnu_cxx::_Lock_policy)2> const&) (__r=..., this=0x55555a785cd8) at /usr/include/c++/9/bits/shared_ptr_base.h:843
#3  std::__weak_ptr<openvslam::data::keyframe, (__gnu_cxx::_Lock_policy)2>::operator=(std::__weak_ptr<openvslam::data::keyframe, (__gnu_cxx::_Lock_policy)2> const&) (__r=..., this=0x55555a785cd0) at /usr/include/c++/9/bits/shared_ptr_base.h:1577
#4  std::weak_ptr<openvslam::data::keyframe>::operator=(std::weak_ptr<openvslam::data::keyframe> const&) (__r=<error reading variable: Cannot access memory at address 0xbfde878c8ffe612f>, this=0x55555a785cd0) at /usr/include/c++/9/bits/shared_ptr.h:531
#5  openvslam::data::landmark::erase_observation(std::shared_ptr<openvslam::data::keyframe> const&) (this=0x55555a785bb0, keyfrm=std::shared_ptr<openvslam::data::keyframe> (use count 1221, weak count 2327) = {...})
    at /home/maverick64/c++_ws/openvslam/src/openvslam/data/landmark.cc:77
#6  0x00007ffff7bf34c7 in openvslam::optimize::local_bundle_adjuster::optimize(std::shared_ptr<openvslam::data::keyframe> const&, bool*) const
    (this=0x555555be63b0, curr_keyfrm=std::shared_ptr<openvslam::data::keyframe> (use count 1221, weak count 2327) = {...}, force_stop_flag=force_stop_flag@entry=0x555555be5e38) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#7  0x00007ffff7b2962e in openvslam::mapping_module::mapping_with_new_keyframe() (this=this@entry=0x555555be5d40) at /usr/include/c++/9/bits/unique_ptr.h:360
#8  0x00007ffff7b29d14 in openvslam::mapping_module::run() (this=0x555555be5d40) at /home/maverick64/c++_ws/openvslam/src/openvslam/mapping_module.cc:105
#9  0x00007ffff6ba2de4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff7283609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#11 0x00007ffff6890293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
based on the gdb message on line #4  dynamic memory allocated by landmark has an invalid memory address. I carefully assume that the wrong number of observations counts. Are there any recent modification histories using the class `openvslam::data::landmark`?

ymd-stella, 2021-12-27:
@Sang-Beom-Woo 
Which data set are you using?

>  Are there any recent modification histories using the class openvslam::data::landmark?

I've only recently started using landmarks in shared_ptr. There may be some mistakes.

Sang-Beom-Woo, 2021-12-27:
I am currently using realsense D435i camera online by sending depth and color images via ROS2 topic. I am using it for large-scale environments and the mapping process usually takes 30min. The run time error appears around 10min after I run the code. I think the conventional offline dataset(approx. 5~7min) might finish the sequence before the run time error appears.

ymd-stella, 2021-12-27:
Did relocalization or loop closing occur?

Sang-Beom-Woo, 2021-12-27:
Yes. I've got multiple loop closures. Also, when the motion blur occurs because of the rough maneuver for while, it successfully relocalize camera motion afterward if there are recognizable features detected. Just before the run time error occurs, everything works ok.
I've downgraded my code, just for now, with commit no. b30a9ccf60aa2f4f57ae7e28b50ed3e81609a068. And it works without error so far.

ymd-stella, 2021-12-27:
It would be helpful if you could check if the same error occurs when the following parameters are set.

```yaml
Tracking:
  enable_auto_relocalization: false
LoopDetector:
  enabled: false
```

abylikhsanov, 2021-12-27:
Yes I had the same problem, it only occurs during the loop closures. I only get this crash on the first loop closure and it appears randomly, quite hard to catch. I am in debugging more but cannot see any exceptions risen

ymd-stella, 2021-12-27:
CI uses a dataset for testing loop closing. Do you get the same error with this?

https://github.com/OpenVSLAM-Community/openvslam/blob/577cb04d9245aa68b0905212293a802292d01787/.github/workflows/main.yml#L102-L118

ymd-stella, 2021-12-27:
What is your OS and g++ version?

Sang-Beom-Woo, 2021-12-28:
Test environment 

[intel NUC core i7 10gen with 16GB RAM] 
OS                   : Linux Mint 20.2
g++ version     : 9.3.0
ROS2 distro.   : foxy

[RYZEN 7 5800H with 32GB RAM]
OS                  : ubuntu 20.04 LTS
g++ version    : 9.3.0
ROS2 distro.  : foxy

Sang-Beom-Woo, 2021-12-28:
Suggested `openvslam_test_dataset` work fine in both of my environments. I ran the code in my dataset 3 times each. During 2 loop closures in the dataset, errors did not occur. I think this dataset does not have enough loop closures to observe the error. I will let you guys know about the following setting in my mapping site when I finish testing.
```
Tracking:
  enable_auto_relocalization: false
LoopDetector:
  enabled: false
```

ymd-stella, 2021-12-28:
> I only get this crash on the first loop closure

> I think this dataset does not have enough loop closures to observe the error.

Isn't this a contradiction?

Sang-Beom-Woo, 2021-12-28:
I am not sure right now, because the run-time error in my computer behaved differently from what abylikhsanov(https://github.com/OpenVSLAM-Community/openvslam/issues/237#issuecomment-1001657187) described 
The code ran in my environment closed loop several times until the software crashes.

To @abylikhsanov: Would you mind if I ask you about the description of the datasets tested and the software environment only if it does not bother you?

jamestkpoon, 2022-02-17:
I am also experiencing this issue when running from a [Dockerfile](https://github.com/OpenVSLAM-Community/openvslam_ros/blob/galactic-devel/Dockerfile) Stack trace is as follows:
```
*** Aborted at 1645101856 (unix time) try "date -d @1645101856" if you are using GNU date ***
PC: @     0x7f28d5257722 openvslam::data::landmark::erase_observation()
*** SIGSEGV (@0x0) received by PID 1236 (TID 0x7f28bbfff700) from PID 0; stack trace: ***
    @     0x7f28d5486631 (unknown)
    @     0x7f28d4d133c0 (unknown)
    @     0x7f28d5257722 openvslam::data::landmark::erase_observation()
    @     0x7f28d52f6aa8 openvslam::optimize::local_bundle_adjuster::optimize()
    @     0x7f28d52148b5 openvslam::mapping_module::mapping_with_new_keyframe()
    @     0x7f28d5216930 openvslam::mapping_module::run()
    @     0x7f28d45efde4 (unknown)
    @     0x7f28d4d07609 start_thread
    @     0x7f28d442e293 clone
```

My dataset is grayscale stereo from a ZED2  log from a robot driving around a loop once. The crash occurs at seemingly random moments during playback which I would not consider candidates for loop closure (given the data in the log I would only expect a single loop closure event near the end), so perhaps it is during relocalization following tracking loss. 
Unfortunately adding 
```
Tracking:
  enable_auto_relocalization: false
LoopDetector:
  enabled: false
```
does not seem to remedy this.

ymd-stella, 2022-02-17:
@jamestkpoon 
What is the commit ID of the code you used?

jamestkpoon, 2022-02-17:
@ymd-stella was on "galactic-devel", but can confirm I get the same behavior with the latest commit on "ros2" fdbd287d80a3a5cc4ac1a143dc937adb4697b120

ymd-stella, 2022-02-17:
@jamestkpoon What is the commit ID of openvslam?

jamestkpoon, 2022-02-17:
@ymd-stella 
958b144f2b1c54b1fa40e881fcc8dbd1f3bb9f3b

ymd-stella, 2022-02-18:
If the settings in https://github.com/OpenVSLAM-Community/openvslam/issues/237#issuecomment-1042925354 are applied correctly, the error will occur even if LoopClosing is disabled and Relocalization is disabled. I am also wondering if it occurs even if Viewer is disabled.
@jamestkpoon Could you please try https://github.com/OpenVSLAM-Community/openvslam_ros/blob/ros2/Dockerfile.cui?

jamestkpoon, 2022-02-18:
@ymd-stella Firstly thank you very much for your assistance with this issue, I really appreciate the effort.
With the log from the robot, the UI-less Dockerfile.cui also results in crashes. However this may actually be an issue with the log itself; throughout the log there is a large circular artifact from a poorly fitted visor which would likely derail performance e.g.:
[image]

Testing with data obtained by waving the camera around freehand works fine even with the UI enabled (as per the original Dockerfile). Will test more extensively next week when possible. Apologies in advance if my additions to this issue were erroneous.

ymd-stella, 2022-02-18:
I'm wondering if the error will occur if you don't use openvslam_ros, but use https://github.com/mlaiacker/rosbag2video to convert it to a video and enter it directly into [run_video_slam](https://openvslam-community.readthedocs.io/en/latest/simple_tutorial.html#tracking-and-mapping).

ymd-stella, 2022-02-19:
I have just reproduced it myself. The dataset is euroc/MH_01, and I used openvslam with PangolinViewer.

PR #257 (2022-02-19): Fix landmark.cc (empty observations_)
Fixed #237.
closed 2022-02-19 by ymd-stella