stella-cv/stella_vslam #249: OpenVSLAM is derivative work of ORB-SLAM2 and must use GPLv3 license

jdtardos, 2022-01-08:
Dear OpenVSLAM community,

I am Prof. Juan D. Tardos, leader of the ORB-SLAM team at University of Zaragoza, Spain. As you know, ORB-SLAM, ORB-SLAM2 and ORB-SLAM3 have been licensed under GPLv3 for their free use in open-source projects. In parallel, our University, as owner of the software, is selling commercial licenses to companies all around the world that want to use our software in closed-source products. This gives the best of both worlds, research labs can use it for free, and companies making money with our software have to share some of their profits with our University. As the software authors, that is exactly our will.

Recently we got aware of the existence of OpenVSLAM, thanks to a big company that was using it. After the termination of the oficial OpenVSLAM release, the company analyzed the problem and concluded that OpenVSLAM is in fact derivative work of ORB-SLAM2, and to be on the legal side, they decided to buy a license of ORB-SLAM2 from our University. 

As authors of the software, and together with the University's IP office, we also have analyzed in detail the issue and our conclusions are:

1. There is no doubt that the original OpenVSLAM paper incurred in academic misconduct. They are using all the ideas in ORB-SLAM and ORB-SLAM2 papers without giving proper credit to the original authors. Some examples: using three threads for tracking mapping an place recognition, using the same ORB features for everything, fast keyframe insertion and removal of redundant keyframes later, building a covisibility graph and an essential graph, using local bundle adjustment based on the covisibility graph, doing pose-graph for loop closing using the essential graph, etc. Their paper only contains a generic citation to ORB-SLAM2, together with other SLAM papers that are totally unrelated to OpenVSLAM. So it's obvious that the OpenVSLAM authors were hiding the fact that their work was heavily based on ORB-SLAM. This went undetected as it was published in a software conference, and not in a robotics conference full of SLAM experts.

2. There is no doubt that OpenVSLAM software is derivative work of ORB-SLAM2. Just to be clear: we have compiled a significant amount of evidence that OpenVSLAM software was written based on ORB-SLAM2 source code, and not only based on the ORB-SLAM and ORB-SLAM2 papers. This evidence comes in the form of same graphical interface, many idiosyncratic code sections (things that could be programmed in tens of different ways, but are programmed in the same exact way as in ORB-SLAM2) and a few obscure bugs that are reproduced in the same exact way, some of them found by members of the community (thanks!!). Sorry, but the legal team in our IP office does not allow us to give more details because the information we have collected can be used as evidence in case of legal claims.  Also here, it's pretty obvious that the authors of OpenVSLAM were hiding the fact that their software is derivative work from ORB-SLAM2. In fact, when their organization decided to cose the original repository, they suggested that to be legally safe, you should consider OpenVSLAM as derivative work of ORB-SLAM2.

For these reasons, we kindly ask you to change the license of this repository to GPLv3. It is totally false that you cannot do it, because people that contributed in good faith to  the repository did that assuming a BSD license, and there is nothing in the BSD license that precludes to redistribute it under GPLv3. 

But the oposite is true, the GPLv3 license of ORB-SLAM2 does no allow to redistribute OpenVSLAM, being derivative work of ORB-SLAM2, under BSD. If you continue under BSD, you would be supporting academic misconduct and open software misconduct, and you would be putting good faith users of your repository in risk of facing legal claims. 

Yours sincerely,
Juan D. Tardos

ymd-stella, 2022-01-08:
Thank you for your comment.
I have understood your intentions. I have some questions about the conclusions.

> 1. There is no doubt that the original OpenVSLAM paper incurred in academic misconduct. ...

It is a matter between ideas, and has nothing to do with licenses that protect expression. If necessary, it would be better to contact the original authors of OpenVSLAM.

> 2. There is no doubt that OpenVSLAM software is derivative work of ORB-SLAM2. ...

Obviously, there is no need to make all of OpenVSLAM GPL, and it should only be necessary to address the derivative work part. For example, the graphical interface using Pangolin can be extracted to an external repository and made GPL. Would you be able to provide the problematic part?

jdtardos, 2022-01-08:
> Thank you for your comment. I have understood your intentions. I have some questions about the conclusions.
>
>
> It is a matter between ideas, and has nothing to do with licenses that protect [...]

That's why I separated it in two different points. This simply proofs unethical behavior in the OpenVSLAM paper and repository. They were copying our ideas and software and trying to hide it.

> Obviously, there is no need to make all of OpenVSLAM GPL, and it should only be necessary to address the derivative work part. For example, the graphical interface using Pangolin can be extracted to [...]

Sorry, the derivative part is most of the software, the interface is just one example, and the idiosyncratic sections and bugs are just evidence that we have collected for our legal team. Except for a few added functionalities (some new camera types) OpenVSLAM is essentially a refactored version of ORB-SLAM2 source code, that was reprogrammed using ORB-SLAM2 source code from the beginning. It was never independently programmed from scratch, as the authors pretended and some of the community members seem to believe. Modifying it further or as you say "eliminating the similarities" would change nothing, the result will still be derivative work of ORB-SLAM2, as derivative work is transitive: if B is derived from A and C is derived from B, then C is derived from A. Please, read the GPL license or ask your legal team: if A is GPL, both B and C must be GPL. Some legal teams in US companies just forbid their engineers to even read any GPL source code to avoid any contamination. If you want a clean system, you need to start from scratch, without reading other's GPL software. 

Here is my summary of the current situation:
- The authors of the original ideas and software (ORB-SLAM2) don't want companies to use their software in closed source products for free, and they chose to open-source it under GPLv3. 
- The OpenVSLAM paper incurred in academic misconduct, and their software violated ORB-SLAM2 GPLv3 license.  
- The authors of OpenVSLAM have recognized the problem by erasing their repository and warning potential users. 
- They have not tried to remove the conflicting parts, because they know they are everywhere, and it would still violate the GPLv3 license. Again: ask your legal team.
- The only legal solution is recognizing ORB-SLAM2 influence into OpenVSLAM and changing it to GPLv3. 

And here is the ethical issue: why would this community remove the proofs of the offense and give the software to be used for free in closed-source products, going further against the ORB-SLAM2 GPLv3 license? Please, think on it.

Yours sincerely,
Juan D. Tardos

ymd-stella, 2022-01-09:
First of all, it is important to note that ideas (algorithms and architectures, etc.) are not protected by the GPL. Are you sure you're not misunderstanding it?

In this view, most of the things quoted from ORB-SLAM2 are ideas, and only some expressions are stolen.

> If you want a clean system, you need to start from scratch, without reading other's GPL software.

This is strictly a misunderstanding. Do you know about [Computer Associates International, Inc. v. Altai, Inc.](https://en.wikipedia.org/wiki/Computer_Associates_International,_Inc._v._Altai,_Inc.)?
There is precedent for eliminating areas of similarity through [clean room design](https://en.wikipedia.org/wiki/Clean_room_design).

ymd-stella, 2022-01-09:
An additional case: mimemagic has reverted from GPL to MIT by removing the GPL-violating parts.

https://github.com/jellybob/mimemagic/pull/3

ymd-stella, 2022-01-09:
Removed the code from the default branch and added a note in README.md that past openvslam must be used under GPLv3.

I am also working on the Unrelease process.
https://github.com/ros/rosdistro/pull/31716.

ymd-stella, 2022-01-09:
Once openvslam is GPLv3 and Unrelease is completed, can you publish the problematic part?

jdtardos, 2022-01-09:
> I sincerely hope that the intentions of the authors of ORB-SLAM2 and the original authors of OpenVSLAM will be respected fairly.

Seriously?.  The intentions of us as ORB-SLAM2 authors are extremely clear:
- Give proper credit to all the intellectual contributions to our work. Please read our papers.
- Respect the license of all software we have used in ORB-SLAM2.
- Make our software free for open-source projects under GPLv3.
- Allow our University (by the way, a university paid by the taxpayers in Spain) to obtain economical returns from the closed-source uses of our work.

The intentions of OpenVSLAM authors seem to be:
- Hide the intellectual contributions of ORB-SLAM2 to their work. Please, read their paper.
- Violate the GPLv3 license of ORB-SLAM2, hiding that it is derivative work. Please, read their software.
- Give our work for free to closed-source companies under BSD.
- To date, no private or public apology has been given to us.
Sincerely, I see nothing to be respected here.

If you cannot see the difference, sorry, but I will stop any further communication.

ymd-stella, 2022-01-09:
Have you seen https://github.com/OpenVSLAM-Community/openvslam/blob/in-discussion/README.md?

ymd-stella, 2022-01-09:
You are talking about the past, and I am talking about the future.

ymd-stella, 2022-01-09:
It appears to me that you are violating rights by bringing up other reasons. Shouldn't we only be talking about the rights protected by GPLv3 here? I think leaving this discussion as it is will be a large barrier to future OSS development of Visual SLAM.

ymd-stella, 2022-01-09:
> - Hide the intellectual contributions of ORB-SLAM2 to their work. Please, read their paper.

I have already read it. I think the responsibility lies with the authors. I think you should contact the authors if necessary.

> - Violate the GPLv3 license of ORB-SLAM2, hiding that it is derivative work. Please, read their software.
> - Give our work for free to closed-source companies under BSD.

Currently, OpenVSLAM is published under GPLv3.

> To date, no private or public apology has been given to us.

You should contact the authors as well. Not us.

ymd-stella, 2022-01-09:
If you refuse to provide information, I will republish OpenVSLAM under the BSD license after I have sufficiently removed the suspect code on my own.

SteveMacenski, 2022-01-11:
The licenses are for the specific code produced, not the ideas that the code represents. No one has claimed that OpenVSLAM doesn't use some (many) of the same ideas as ORB-SLAM, but using the same concepts is not the same thing as the same literal lines of software. The OpenVSLAM paper openly cites ORB-SLAM3, among others, as direct sources of work it builds on. I don't see the issue.

hitdshu, 2022-01-17:
Actually, in China, almost every robotics company uses ORB-SLAM, and I think only very few of them are paying for it.

But for OpenVSLAM, I don't think there is a problem in its paper. They've clearly cited ORB-SLAM and their contributions are clearly indicated.

devrite, 2022-01-18:
The only way OpenVSLAM can be non-GPL is if any claim by Prof. Tardos and its group is not valid due to the reasons you identified:

* The way to implement it given all papers at the time could be considered common knowledge or the standard/naive taught way to do it (i.e. by following design rules/patterns, data norms like when designing via UML or ER diagrams, etc.).

Otherwise any code similarity is basically (even without GPL) first protected by copyright and would be considered plagiarism if too frequent or too many.

Prof. Tardos claims that they have evidence even bugs were the same. Consider an exam, test, paper, home work. If find many identical errors or similar formulations, though given spoken language gives you more freedom than code, your evaluator would consider these parts an uncited copy. If it was not a bug in the paper itself, this is very suspicious, especially if it is not just one. Also you identified code parts that were too similar yourself.

For OpenVSLAM, which in terms of architecture, code quality, ... it did improve on ORB(2)-SLAM this means if there are enough affected parts these parts are protected by copyright and since their license states (GPL) when it is legal to copy and modify parts (by maintaining GPL) it affects any change after the initial publication or problematic version of OpenVSLAM made to it. GPL just copies over to successor change.

While in terms of efficiency and not reinventing the wheel twice terms and do unnecessary work I get your sentiment and that it would not really much differ to the end result, but our laws and licenses define the rules are quite clear once a claim is valid. It might not make much sense or cause frustration. But the only way, given the claims hold, to publish a similar package with another license is by completely re-implementing it in such a way that it can't be considered a copy of ORB(2)-SLAM and frustratingly nor of OpenVSLAM. Again if the claims hold.

Meaning if there had not been any published OpenVSLAM repo and the claims identified would have been fixed before publishing it (=initial commit), you would be right it would not need to be protected by GPL. But this is not the case, again given the issues and claims identified or sufficient. Which I am not making any comment on.

The question to be solved is as stated above:

* Are the problematic parts novel and unique ways to implement it (not standard reached by following common rules/knowledge or suggested hinted by SoTA or even the paper itself)
* Were problematic parts present in any published version of OpenVSLAM.

If both answers are yes. Prof. Tardos is right. Removing does not solve it. The first part is not easy to asses and usually leads to a lot of work, as you would have to bring up any rule to derive it any other SLAM or similar algorithm that does it in this way at that time or hints in their paper. I do not want to discourage you nor is it the obligation of Prof. Tardos to work for you.

Basically how confident are you about your analysis (not only the current but any version in the histroy of OpenVSLAM).

In terms of academic misconduct. I am not sure myself. I would have to check, by initially browsing the paper I would say the separated from Sota and have written what their contributions were. Although not explicitly stating that they are otherwise identical. But the papers I read often state it like that. So, at least Idk enoguh to claim that. But copyright issues here is not important anyway.

ymd-stella, 2022-01-18:
@devrite Thanks for your comments.

> - The way to implement it given all papers at the time could be considered common knowledge or the standard/naive taught way to do it (i.e. by following design rules/patterns, data norms like when [...]

First of all, I accept the existence of plagiarized parts, but I deny the claim that all of OpenVSLAM can be considered plagiarized.

> it is legal to copy and modify parts (by maintaining GPL) it affects any change after the initial publication or problematic version of OpenVSLAM made to it. GPL just copies over to successor change.

> Removing does not solve it.

This is a misunderstanding. Do you know about [Computer Associates International, Inc. v. Altai, Inc.](https://en.wikipedia.org/wiki/Computer_Associates_International,_Inc._v._Altai,_Inc.)?
There is precedent for eliminating areas of similarity through [clean room design].
What are your thoughts on this?

> For OpenVSLAM, which in terms of architecture, code quality, ... it did improve on ORB(2)-SLAM this means if there are enough affected parts these parts are protected by copyright.

This is a misunderstanding. Copyright only protects expression, not ideas. Just being affected by it is not considered plagiarism. See https://copyleft.org/guide/comprehensive-gpl-guidech5.html. Furthermore, it is recommended that you also read [Merger doctrine](https://en.wikipedia.org/wiki/Idea%E2%80%93expression_distinction#Merger_doctrine).

> The first part is not easy to asses and usually leads to a lot of work, as you would have to bring up any rule to derive it any other SLAM or similar algorithm that does it in this way at that time or hints in their paper.

It's not logical, and I don't know how well I understand your point, but I don't think I need to do that.
In his first comment, he stated the following:

> This evidence comes in the form of same graphical interface, many idiosyncratic code sections (things that could be programmed in tens of different ways, but are programmed in the same exact way as in ORB-SLAM2) and a few obscure bugs that are reproduced in the same exact way

According to this, they clearly show that the target code section can be implemented in tens of different ways. This means that they can be removed by implementing them in a different way.

devrite, 2022-01-19:
Sure, but even in the case you posted they were only allowed to remove it as it could be seen as a different separate entitity (CA Scheduler), and it was only US law that decided it. While they are in Spain and theory could sue anywhere.

While here this may be true for separte modules, libs or programa than can be extracted I doubt it would be true if these issues reside in the core part. For example while io, cam models,... might be seen separate I doubt any issue in the core lib will be.

For example the Ca-Scheduler issue of your case could be seen equal to the role of g2o. You identified issues in core structures yourself. They, identified the same bugs, at the same locations.

The problem is: Is this core part, that makes OpenVSLAM and which can not be seen as separate entitities at the time of publishing, too similar. 

The core part is split in tracking, loop closure, map data, feature extraction and matching and the overall architecture. But some (loop closure, map) they are tightly coupled published in a sense that any change in data, structure will affect a lot of other parts. Also if issues were found in many entities not just one it could be seen as infringement.

Meaning if the way they solved integrating the edges in the g2o graph us not complitely derivable from the paper to arrive at such similarities this module could be seen as too similar.

Saying if you take any SW engineeer

devrite, 2022-01-19:
For me as bystander there is currently no real confidence whether it is BSD or GPL. As your view may not be shared by actual law experts or that your identified units of abstraction are not ones others would argue or if the number of affected parts or key core parts are insignificant. Also you would need to check the original code, which may have many more problematic code parts and were just refactored later.

In my point of view you only GPL currently gives you any certainty. You may post a note along side it that, this issue is not resolved and once certainly is the right is maintained to switch it back. But without consulting experts (sample n=1 or n=3 in this issue) our discussion is not really well founded. Law needs usually many different point of views to see if this is the consensus or just our biased or incomplete views/understanding.

AlejandroSilvestri, 2022-01-23:
Hi all,
I want to share my opinion and some objective observations to enrich this very important debate, and as far as possible bring positions closer together to reach an agreement. That is why I want to start by exploring the zone of possible agreement by interpreting each one’s interests.  Without the intention of taking part, we agree we can have different opinions, and if I am wrong about something, I would appreciate it if you pointed it out to me.

Dr. Juan Tardós, whom I respect and admire, made it very clear the interest of his institution: that OpenVSLAM adopt the GPL license; and the reason: you will be able to use ORB-SLAM2 for free under the GPL license, or agree to a license with the university for profit. He also asks acknowledgement from OpenVSLAM authors, but this is not the place to do so as the authors do not participate here.

I want to emphasize the dedication and zeal of ymd-stella in carrying out this project and this community. I know ymd-stella interests from other threads: keeping a VSLAM system under the BSD license, without the GPL obligations. To do this, ymd-stella requests help to identify the parts that violate the ORB-SLAM2 license to rewrite them. And one of the arguments ymd-stella makes for not giving up BSD is that presumably many contributors worked on this project precisely because the license is BSD, and they wouldn't have done so if it were GPL. At the end of the day, this community is looking for a free and open VSLAM that can be used commercially without paying royalties.

Dr. Tardós declares they have collected a long list of evidence showing that OpenVSLAM is a derivative work of ORB-SLAM2, for which it is appropriate to adopt the GPL license. Being a legal matter, it is logical that it is in the hands of a lawyer and that he is not allowed to share more details, against ytm-stella’s interests of patching the code to avoid GPL.

These are my statements that I will substantiate later:
- OpenVSLAM is a derivative project of ORB-SLAM2 in practical terms, not meaning it is so in legal terms
- This project can give credit to ORB-SLAM2
- OpenVSLAM's structure is disturbingly similar to ORB-SLAM2, and it wouldn't pass an AFC test

I'm not new to the scene, I reverse-engineered ORB-SLAM and ORB-SLAM2 years ago by documenting the code, which allowed me to get familiar with the code. I have also studied others, such as LSD-SLAM, and I have to point out that although they seek the same thing, the way of approaching it is completely different.

More recently I tried (and abandoned) a rewrite of ORB-SLAM2, and exploring improvements found many ways to formulate the algorithm for more performant, clearer and more maintainable code. I abandoned the project when OpenVSLAM was released: someone had already done better. Studying its code I found innumerable parallels with that of ORB-SLAM2, and was surprised that it was hardly mentioned in the paper.

OpenVSLAM looks exactly like a rewrite of ORB-SLAM2 with:
- better programming practices
- a more modular and tidy class structure
- clearer variable names
- cleaner code thanks to the use of modern C++
- parallelism with OpenMP
- alignment
- shared pointers adoption

and more.  It seems to be the result of the “clean room” copy method, which would explain the similarities while avoiding being legally considered a “derivative”.  At this point, I have no doubt OpenVSLAM is closely related to ORB-SLAM2 in practical terms, but not in legal terms of licensing.

As Dr. Tardós indicates, OpenVSLAM implements all the ideas of the ORB-SLAM2 paper; this is evidenced in the OpenVSLAM code and does not support any other interpretation. But this does not mean recognizing the GPL license, since the paper is public and therefore the ideas it presents are not subject to the GPL license. I think it would be fair to mention in README.md the paper's contribution to this project.

Finally, an exhaustive comparison of both codes is a huge job that nobody wants to do. From the description of the copyleft AFC, I understand (this is my opinion) that OpenVSLAM hosts a huge amount of the “kernels of creative expression” from ORB-SLAM2. For example:

- The first obvious step in the process is to extract ORB descriptors from the image. Both ORB-SLAM2 and OpenVSLAM walk the same strange path before doing so (under different names):
  - SLAM.feed_monocular_frame(im)
  - tracker_->track_monocular_image(im)
  - new frame(im), where the image is finally processed
- the Frame class process the image in its constructor: this is a very arbitrary and odd programming decision, the same in both projects, and it's not the obvious or self-evident way to do things
- track_monocular_image() constructs frame() in one of two different ways, depending on whether they are in initialization or tracking
- the main sequencer finite state machine is located in tracker, and is the same in both projects
- FAST keypoints are filtered by OctTree or QuadTree (same process, different name), when it is neither the most obvious nor the best way to do it

There are countless similarities that are not casual nor the most common way of dealing with them. I believe Dr. Tardós when he says that he has a lot of evidence showing that OpenVSLAM is a derivative work of ORB-SLAM2.  More importantly, these similarities aren’t modular, they go to the core of OpenVSLAM.

In conclusion, adopting the GPL for OpenVSLAM automatically moves to GPL all added work, which is a lot and significant, and goes against the spirit of this project.  On the other hand, it is possible to modify OpenVSLAM to no longer be a derivative work., and It requires more effort than the authors did in the first, but this work is already in progress.  This way OpenVSLAM will be based on ORB-SLAM2 paper, while not being a derivative work from ORB-SLAM2 GPL code.

devrite, 2022-01-23:
The problem is if your conclusion is really ght that it indicates to be derivative work and is to be considered so then the work in progress is also a modification and derived of Openvslam. Especially since you track all your changes. This way you can never do a clean room copy as you already mentioned dies not affect an exchangeable module. 

Meaning if GPL is to be adopted and the project wants to be GPL free it may keep the name but must not itself be considered a copy/derivative of any Openvslam version that should have been licensed via GPL. Meaning if you just start by cooying the interfaces,... if that were not to pass afc would also need to adapt GPL. If the project wants to drop that it is better to start anew without looking at the code.

ymd-stella, 2022-01-23:
@devrite Your comment has many typos and does not convey the message, so it would be helpful if you could reread it carefully and correct it.

devrite, 2022-01-23:
Updated: Sorry sometimes auto-correct adds or drops stuff I did not notice.

ymd-stella, 2022-01-23:
Whether or not OpenVSLAM can be made non-GPL depends on whether you think GPL was mixed into OpenVSLAM under non-GPL, or whether you think OpenVSLAM was originally under GPL.  I am of the former opinion; devrite seems to be of the latter opinion.

To date, I have done comparisons of all ORB-SLAM2 code and corresponding similar OpenVSLAM code. It is my opinion that at least some of the code was created by clean room design or similar methods. Some of the code is completely unique. Rather than release them under GPLv3 against their authors' intentions, I would choose to remove the GPLv3 infringing code even if it no longer works as a system.This is because the community's contribution so far has been towards the BSD-licensed library. In addition, I don't want to expose the new Visual SLAM library that will be created in the future to the threat of GPLv3.

ymd-stella, 2022-01-29:
Dear @jdtardos,
I have replaced the code that is suspected of violating the GPL. See https://github.com/OpenVSLAM-Community/openvslam/pull/252. If there are any remaining problems, please point them out.

ymd-stella, 2022-02-06:
Version 0.3.0 with the fixes applied will be released again under the BSD license. If there are any issues with the license, please point them out to me and I'll address them as soon as possible.

ymd-stella, 2023-02-11:
One year has passed with no updates. I will close it. If you find any issues, please create another Issue.

closed 2023-02-11 by ymd-stella