stella-cv/stella_vslam #250: CPU usage goes up to 100% if the viewer is removed when the slam is in localization mode

Valts-M, 2022-01-27:
Since for my purposes I won't be needing to be able to see what the slam is doing I tried to remove the viewer to save on processing time. I replaced the viewer.run() call with a simple std::cin.get() and a SLAM.request_terminate() right after it so that I can stop the process by imputing anything to the terminal. Everything else about the structure of the program stayed the same. The frames still get fed in a separate  thread. I noticed that now the CPU usage went from about 40% to 100% with this simple change, but only if the slam was in localization mode. I tried to figure out what was the cause but didn't notice anything that could possibly cause this spike in CPU usage. I've also made sure that I have built a Release version.

To make sure it wasn't a mistake in my own code somewhere I tried to make these changes in run_video_slam and run_video_localization. I ran it with the aist data set and witnessed the same behavior. For reference I am running the slam on a laptop with a  i7-10750H and 16 GB of RAM.

ymd-stella, 2022-01-27:
Do you set the option `--no-sleep`? With this option, no sleep is performed for tracking, which will result in high CPU usage.

Valts-M, 2022-01-28:
I tried both with and without no sleep. In either case the CPU usage jumped to 100%.

ymd-stella, 2022-01-28:
I was able to reduce the CPU usage by removing `--no-sleep` and specifying `--frame-skip 3`.

If you want to achieve optimal frame drop for realtime processing, I recommend using openvslam_ros.

ymd-stella, 2022-01-28:
In addition, if you want to disable the viewer, it is recommended to specify `-DUSE_PANGOLIN_VIEWER=OFF` as the build option.

closed 2022-01-28 by ymd-stella