stella-cv/stella_vslam #271: cv::KeyPoint may not be saving and loading as intended

ymd-stella, 2022-04-03:
## Describe the bug

When saving maps to sqlite3, cv::KeyPoint may not be saving and loading as intended. The following warning has been issued.

```
/openvslam/src/openvslam/data/map_database.cc:538:76: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class cv::KeyPoint' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
```

## Environment

- CPU: Intel
- OS: Ubuntu18.04
- Commit id: 64f4f63b187a3f09927fe7510e4a461271e9d8c3
- Install procedure: docker

## Additional context

Related to https://github.com/stella-cv/stella_vslam/pull/269.

BBin1103, 2022-05-20:
A bug in map_database::register_keyframe, bearings.size() is the double of num_keypts. Possible repair methods are as follows.

from
```
    // bearings
    auto bearings = eigen_alloc_vector<Vec3_t>(num_keypts);
    assert(bearings.size() == num_keypts);
    camera->convert_keypoints_to_bearings(undist_keypts, bearings);
```
to
```
    // bearings
    auto bearings = eigen_alloc_vector<Vec3_t>(num_keypts);
    bearings.clear();
    camera->convert_keypoints_to_bearings(undist_keypts, bearings);
    assert(bearings.size() == num_keypts);
```

ymd-stella, 2022-05-20:
@BBin1103 Thanks for the report, but please create another issue because it is irrelevant to this issue.

BBin1103, 2022-05-20:
> @BBin1103 Thanks for the report, but please create another issue because it is irrelevant to this issue.

ok, [#308](https://github.com/stella-cv/stella_vslam/issues/308)