stella-cv/stella_vslam #291: Create initial frame with gravity aligment from IMU

BioR0bot, 2022-04-25:
It would be very convenient to create a top view map using gravity alignment.

closed 2022-04-25 by ymd-stella

BioR0bot, 2022-04-25:
This feature is not difficult to implement, is it?

ymd-stella, 2022-04-25:
It would not be difficult to implement it yourself; I think it is beyond the scope of the Visual SLAM library's main responsibilities. I do not plan to include post-processing of the completed maps in this repository.

ymd-stella, 2022-04-25:
It is important to note that simply aligning the first keyframe may result in other keyframes having different gravity directions due to errors.

BioR0bot, 2022-04-25:
I saw that you tried to add work with sensors? Where can I get a piece of code from you that solves this problem?

BioR0bot, 2022-04-25:
The map is initialized as far as I know with the first frame. It has an identity rotation matrix. To rotate the entire system, you need a transformation matrix from the sensor and a transition matrix from the camera to the sensor

BioR0bot, 2022-04-25:
> It is important to note that simply aligning the first keyframe may result in other keyframes having different gravity directions due to errors. (Aligning the whole map is dealt with at #235.)

code is in wip branch?

ymd-stella, 2022-04-25:
Yes.

BioR0bot, 2022-04-25:
> Yes.

How can I access it?

ymd-stella, 2022-04-25:
https://github.com/ymd-stella/openvslam-community/tree/wip/gba

BioR0bot, 2022-04-25:
> https://github.com/ymd-stella/openvslam-community/tree/wip/gba

Thank You very much)