stella-cv/stella_vslam #297: High cpu usage in localization mode，which is three times cpu usage in slam mode.

BBin1103, 2022-05-11:
Thank you for the clean code!

I noticed that cpu usage is abnormally high in localization mode and there are some other github issues report similar things.  After some tests, I found 2 ways to  reduce cpu usage to normal level, about 60% of cpu usage in slam mode.

1. Turn on mapping thread after relocalization success  for a few frames then pause mapping thread.
2. Compile g2o with -DG2O_USE_OPENMP=OFF then recompile stella_vslam.

It seems that g2o openmp works bad in some cases. Hope this helps.

ymd-stella, 2022-05-11:
ref.

- https://github.com/stella-cv/stella_vslam/discussions/73
- https://github.com/stella-cv/stella_vslam/issues/250

ymd-stella, 2022-06-03:
Closing as there seems to be no additional information. If you find out anything, please create another issue.