stella-cv/stella_vslam #307: Preprocessing::min_size does not have hard or soft upper limit on no of ORB Features to extract

mitul93, 2022-05-18:
I saw in #299 that ORB extraction now uses resolution independent parameter (min_size).
During the test from v0.3.3, I noticed slow run-time. 

I think that's because there is no upper bound of number of features to extract.
Can someone confirm this?

ymd-stella, 2022-05-18:
> Preprocessing::min_size does not have hard or soft upper limit on no of ORB Features to extract

There is an upper limit to the number of feature points. The upper limit is the number of regions when the region division is repeated until the size reaches min_size.

ymd-stella, 2022-05-18:
I know that speeds may decrease under certain conditions. Please let us know your benchmarking conditions and results.

ymd-stella, 2022-05-19:
In fact, (if the appropriate value is set for min_size,) https://github.com/stella-cv/stella_vslam/pull/299 only changes the processing time for feature extraction by a few milliseconds, and the largest change in processing time is for pose optimization. For a fundamental solution, I think it is necessary to put an upper limit on the processing time for pose optimization.

closed 2022-05-21 by ymd-stella