stella-cv/stella_vslam #314: Blank Screen in Pangolin Viewer

zovelsanj, 2022-05-30:
## Problem Encountered
When trying to visualize the equirectangular dataset output, Pangolin Viewer (in both Frame Viewer and Map Viewer) is blank. I followed exactly the same steps for [installation](https://stella-cv.readthedocs.io/en/latest/installation.html) and [visualization](https://stella-cv.readthedocs.io/en/latest/simple_tutorial.html) as described in stella_vslam documentation. Building process is error free for all dependencies and the source as well.
[image]

When running the sample plots (`Pangolin/examples/SamplePlot/main.cpp`) of Pangolin the output is visible and perfect like this:
[image]

It seems Pangolin has problem showing output with just stella_vslam. The Pangolin i built was as suggested in the stella_vslam documentaion (from commit ID [ad8b5f8](https://github.com/stevenlovegrove/Pangolin/tree/ad8b5f83222291c51b4800d5a5873b0e90a0cf81).). 

## Environment
- OS: Ubuntu 20.04
- Install procedure: exactly as described in [stella_vslam documentation](https://stella-cv.readthedocs.io/en/latest/installation.html)
- dataset: aist_living_lab_1 (as given in [Simple Tutorial](https://stella-cv.readthedocs.io/en/latest/simple_tutorial.html) of stella_vslam documention, I checked the video inside aist_living_lab_1 directory it plays completely fine.) 

## Additional context
There is also no problem with opengl, its dependencies and graphics drivers.

Can anyone help me with this?

ymd-stella, 2022-05-30:
Try 

```shell
$ export LIBGL_ALWAYS_SOFTWARE=1
```

ymd-stella, 2022-06-03:
The visualizer in the attached image appears to be normal. It seems to be loading an invalid map and doing localization.

ymd-stella, 2022-06-03:
@zovelsanj If you still have questions, please ask them in discussions.

closed 2022-06-03 by ymd-stella