stella-cv/stella_vslam #322: Expose 'tracking_module.cc::feed_frame()'-parameter 'init_retry_thr' to 'conf'-class

McC0dy, 2022-06-13:
## What issue is the feature request related to?

The system resets if it's unable to relocate shortly after initialization this is reasonable behavior. However, currently the threshold is hard-coded to 5 sec but this value does not fit all applications (mine included),

## Describe the solution you'd like

Expose `init_retry_thr` from `tracking_module::feed_frame()` all the way to the `config` class and the example configuration files' "Tracking Parameters"-section.

### How to achieve this

(From my vague overview of the project - feel free to correct me)
- [ ] In `conf.h/.cc`: Add parameter `init_retry_thr`
- [ ] In `tracking_module.h/.cc`: Through the `tracking_module` constructor expose `init_retry_thr`
- [ ] Optional in `tracking_module.cc`: If `init_retry_thr` is negative then never reset system. Corresponding to `init_retry_thr=infinity`.
- [ ] In `stella-cv/examples`: Update all yaml files to include `init_retry_thr: 5.0`
- [ ] Update documentation accordingly

## Describe alternatives you've considered

Changing the video metadata FPS would allow for indirectly affecting the threshold but could have unintentional consequences.

ymd-stella, 2022-06-13:
> In conf.h/.cc: Add parameter init_retry_thr

Not required.

> In tracking_module.h/.cc: Through the tracking_module constructor expose init_retry_thr

See 
https://github.com/stella-cv/stella_vslam/blob/3f4b160e4eb065cf9a800160e9b38887658cd58f/src/stella_vslam/tracking_module.h#L114-L122

> In stella-cv/examples: Update all yaml files to include init_retry_thr: 5.0

Set the default value to 5.0.

See https://github.com/stella-cv/stella_vslam/blob/3f4b160e4eb065cf9a800160e9b38887658cd58f/src/stella_vslam/tracking_module.cc#L24-L27

> Update documentation accordingly

See 
https://github.com/stella-cv/docs/blob/9cb9458c9ea9e49d8feaa71c278f9872f76a2588/docs/parameters.rst?plain=1#L101-L117

McC0dy, 2022-06-14:
@ymd-stella I've included your feedback in the original description. Thank you.
Did I understand correctly that it's not needed to add `init_retry_thr: 5.0` to the example yaml's?

PR #498 (2023-05-01): Add parameter init_retry_threshold_time
Close #322
closed 2023-05-01 by ymd-stella