stella-cv/stella_vslam #329: Some problems in angle_checker.h

ttaoREtw, 2022-06-23:
Thanks for the great work.  
Just wanna point out that the naming of `histogram_length_` in `angle_checker.h` is kind of confusing.  
It combine two concept
1. The total number of bins
2. The width of a bin

It seems to me it should be separated into two variables as `bin_width_` and `histogram_length_`.  
Also, the angle may be like 720 or larger. This is not handled well now.
File link: https://github.com/stella-cv/stella_vslam/blob/main/src/stella_vslam/match/angle_checker.h#L71.
```c++
template<typename T>
inline void angle_checker<T>::append_delta_angle(float delta_angle, const T& match) {
    // TODO:
    // It should also consider edge cases like -720 or 720
    if (delta_angle < 0.0) {
        delta_angle += 360.0;
    }
    if (360.0 <= delta_angle) {
        delta_angle -= 360.0;
    }
    
    // TODO:
    // Use `histogram_length_` as the width of a bin to get the index. 
    // I would say this is `bin_width_` but not `histogram_length_`
    const auto bin = static_cast<unsigned int>(cvRound(delta_angle * inv_histogram_length_));

    assert(bin < histogram_length_);
    angle_histogram_.at(bin).push_back(match);
}
```

ttaoREtw, 2022-06-23:
I don't know how to change label.  
It is more like an `enhancement` but not `bug`.

ymd-stella, 2022-06-23:
1.

```c++
    // TODO:
    // It should also consider edge cases like -721 or 721
    if (delta_angle < 0.0) {
        delta_angle += 360.0;
    }
    if (360.0 <= delta_angle) {
        delta_angle -= 360.0;
    }
```

If the two angles both range `[0, 360)`, the difference ranges `(-360, 360)`. It seems that the latter decision can be removed. However, it may be necessary to add an assert to avoid giving the wrong input.

It may be a good idea to change the function to one that takes two angles instead of a difference.

2.

```c++
    // TODO:
    // It uses `histogram_length_` as the width of a bin to get the index. 
    // I would say this is `bin_width_` but not `histogram_length_`
    const auto bin = static_cast<unsigned int>(cvRound(delta_angle * inv_histogram_length_));
```

I think it would be correct to calculate the following. What do you think?

```c++
    const auto bin = static_cast<unsigned int>(cvRound(delta_angle / 360.0 * histogram_length_));
```

ttaoREtw, 2022-06-23:
LGTM.  
Thanks!

ttaoREtw, 2022-06-24:
The unit test needs to be revised, too.

https://github.com/stella-cv/stella_vslam/blob/686087db19e56d5a600997554fa80ab06c3fc9b3/test/stella_vslam/match/angle_checker.cc#L12-L20

ymd-stella, 2023-02-11:
angle_checker is no longer required.

closed 2023-02-11 by ymd-stella