stella-cv/stella_vslam #332: Automatic python bindings using Cppyy or similar system

TheBricktop, 2022-06-29:
## What issue is the feature request related to?

I've noticed that in readme.md is a TODO for python bindings, i think that this is pretty easy to do now using numerous automatic binding solutions for c/cpp using python
  

## Describe the solution you'd like

I'll probably try to create python bindings using pybind or cppyy as they are quite mature and used in few other projects, that might even allow to automate the binding process whenever new change will appear in the source code.
### How to achieve this

- [ ] TASK1 Cppyy works by feeding it header file and then linker symbols, then you can use the library itself (hopefully) 
- [ ] TASK2 according to the documentation it goes like that
- [ ] >>> import cppyy
>>> cppyy.include('zlib.h')        # bring in C++ definitions
>>> cppyy.load_library('libz')     # load linker symbols
>>> cppyy.gbl.zlibVersion()        # use a zlib API

## Describe alternatives you've considered

ive found a list of possible binding generators for python and it is here https://github.com/jslee02/awesome-cpp-python-binding-generator

ymd-stella, 2022-06-29:
> I've noticed that in readme.md is a TODO for python bindings,

I left the OpenVSLAM description as is and forgot about it. It is not a high priority for me. I am not sure what demand there is.

TheBricktop, 2022-06-30:
Well im just using the python as a gluecode for opencv, depthai and some networking for me at least the native libs that do the heavy load and a fast modifiable python code is the most optimal approach. And considering that there is a lot people experimenting with computer vision and robotics that do primarily c and python it might be a nice thing to do.

Squiro, 2022-07-07:
Hi,

First of all I'm speaking from ignorance here: I don't know anything about Cppyy. 

But... I'm not sure if an automatic binding tool can generate bindings that work properly for a system like this. I've been doing some work on [Python bindings for StellaVSLAM ](https://github.com/Squiro/StellaVSLAM-Python-bindings) and it's not as straightforward as it would appear. 

For example, to run SLAM _and_ the viewer together in Python, you need to create bindings that release the Python GIL. Otherwise the execution will be blocked. I'm not sure if that's something that Cppyy could infer.

(also, the bindings I've been working on are still a work in progress, there's a lot of room for improvement, I'm still fairly new at this so any help or tips are appreciated)

TheBricktop, 2023-12-28:
Wow thats a lot of work right there. Thank You very much.

zicklag, 2024-03-03:
I can confirm that Python bindings would be useful to me. I'm working on combining Stella VSLAM with OpenCV and OpenMVS for dense reconstruction, and I could connect everything with Python and get my custom camera feed into Stella VSLAM if it had a Python API.