stella-cv/stella_vslam #345: Resources allocated with `new` should be deleted

mitul93, 2022-07-19:
## Describe the bug
Resources allocated with `new` should be deleted.

https://github.com/stella-cv/stella_vslam/blob/4551a4b6dabf5fa1816e8efbda84fff1cda0cdba/src/stella_vslam/optimize/pose_optimizer.cc#L51

https://github.com/stella-cv/stella_vslam/blob/4551a4b6dabf5fa1816e8efbda84fff1cda0cdba/src/stella_vslam/optimize/pose_optimizer.cc#L60

https://github.com/stella-cv/stella_vslam/blob/4551a4b6dabf5fa1816e8efbda84fff1cda0cdba/src/stella_vslam/optimize/pose_optimizer.cc#L48

## To Reproduce
The bug can be checked visually. No need to compile and run the code.

## How to fix?
Use `delete` for resources before returning the function call.

- Hardware: PC
- CPU: i7-12700
- OS: Ubuntu 22.04

ymd-stella, 2022-07-19:
Is it not released inside g2o?

mitul93, 2022-07-20:
you are right `edge`, `vertex` and `algorithm` are released by g2o.
Only `terminateAction` needs to deleted.

Thank you for quick response and fix :)