stella-cv/stella_vslam #351: load_map_db() overrides .yaml features

jamestkpoon, 2022-07-27:
## Describe the bug
When I load a map database with ```SLAM.load_map_database(map_db_path);```, I get the following:
```
load a orb_params "default ORB feature extraction setting" from JSON
```
This seems to happen as part of load_map_database(), as the SLAM system then proceeds to parse keyframes and landmarks etc prior to the line ```[info] startup SLAM system```. The parameters appear to be parsed correctly from the .yaml as the expected values are printed during system(). My current thinking is that the ORB settings used for creating the db end up being included as part of the file.

## To Reproduce

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

config file:

```yaml
# ZED2

#==============#
# Camera Model #
#==============#

Camera:
  name: "ZED2"
  setup: "stereo"
  model: "perspective"

  fx: 526.2252197265625
  fy: 526.2252197265625
  cx: 634.2841186523438
  cy: 348.4753112792969

  k1: 0.0
  k2: 0.0
  p1: 0.0
  p2: 0.0
  k3: 0.0

  fps: 15.0
  cols: 1280
  rows: 720
  focal_x_baseline: 63.147 # fx * 0.12
  depth_threshold: 40

  color_order: "Gray"

#=====================#
# Tracking Parameters #
#=====================#

Preprocessing:
  max_num_keypoints: 2000
  ini_max_num_keypoints: 4000

#================#
# ORB Parameters #
#================#

Feature:
  name: "Altopack Interior Test"
  scale_factor: 1.2
  num_levels: 8
  ini_fast_threshold: 15
  min_fast_threshold: 10

#====================#
# Mapping Parameters #
#====================#

Mapping:
  baseline_dist_thr_ratio: 0.02
  redundant_obs_ratio_thr: 0.9

#========================#
# Initializer Parameters #
#========================#

Initializer:
  num_min_triangulated_pts: 100

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer:
  keyframe_size: 1.2
  keyframe_line_width: 1
  graph_line_width: 2
  point_size: 2
  camera_size: 0.8
  camera_line_width: 3
  viewpoint_x: 0.0
  viewpoint_y: -300
  viewpoint_z: -0.1
  viewpoint_f: 2800
```

## Expected behavior
.yaml parameters are not over-written during the map loading process.

## Environment
- Commit id: 584cff0c84c68d0cd7546d5158a82123fb04d4de
- Install procedure: docker

ymd-stella, 2022-07-27:
> The parameters appear to be parsed correctly from the .yaml as the expected values are printed during system().

When an output exists, it should be reported.

ymd-stella, 2022-07-27:
> Steps to reproduce the behavior:
> 
> Go to '...'
> Click on '....'
> Scroll down to '....'
> See error

Please fill it in always.

jamestkpoon, 2022-07-27:
Thank you, I have added some additional debug log etc.
Can you please confirm if the ORB feature extraction params are being read from the JSON file that's passed into load_map_db()?

ymd-stella, 2022-07-27:
> .yaml parameters are not over-written during the map loading process.

Please add details.
Are you saying that the parameters are not actually used?

ymd-stella, 2022-07-27:
> load a orb_params "default ORB feature extraction setting" from JSON

It means that it will be loaded into **orb_params_database**. The current orb_params is always set from yaml.

closed 2022-07-27 by ymd-stella

ymd-stella, 2022-07-30:
@jamestkpoon Logs and comments added. If you have any questions, please ask.