stella-cv/stella_vslam #356: System looses tracking when a previous map is loaded into the system

youknowimcomingwhenyouhearmehumming, 2022-08-05:
## Describe the bug
You have two videos from the same building. The only differences between the videos is the start and end are from different rooms inside this building. However, the middle part of both videos are from the exact same rooms in the building. 

In one run a map is created and save from one of the videos . 

Then in another run with the other video,  the map is loaded into the system.  The system is able to initialize as normally but right after the initialization, the system looses the feature tracking. When the video gets to the middle part, the system can recognize features from the map and feature tracking is regained. From this point the tracking and mapping module works perfectly so even when the video gets to the different end-part it keeps performing feature tracking and creating new key frames. 

## The value of fixing this bug
Imagine a robot that often have to move around in the same building - if it could use a previously recorded map its navigation would be much more stable.  E.g. if the light in one of the usually corridors is out, the SLAM system would normally lose the tracking if the robot went trough this corridor. However, if the SLAM system had access to a previously map it could recover its location perfectly on the other side of the dark corridor. 

## To Reproduce
In the first run with one of the videos use "SLAM.save_map_database("map_of_stella.msg")" to save the map.

In the second run with the other video use "SLAM.load_map_database("map_of_stella.msg");" to load the map.
(IMPORTANT: do NOT disable mapping module with "SLAM.disable_mapping_module();")
You will then see the following error messages "[info] tracking lost: frame xx". 

If you keep letting it run it will continue with "[info] tracking lost within 5 sec after initialization" and the whole system will reset. This resetting will delete the map you just loaded. You can deactivate the resetting but it does not help to solve the problem. 

## Suggested solution
I noticed a part of the problem is in the tracking_module.cc with the following lines:
    // pass all of the keyframes to the mapping module
    assert(!is_stopped_keyframe_insertion_);
    const auto keyfrms = map_db_->get_all_keyframes();
    for (const auto& keyfrm : keyfrms) {
        mapper_->queue_keyframe(keyfrm);
    }
This code loads all the key-frames from the previously map in a randomized order. 

Therefore, I suggest the following code which makes sure that the key frames are loaded in the correct order:
    // pass all of the keyframes to the mapping module
    assert(!is_stopped_keyframe_insertion_);
    auto keyfrms = map_db_->get_all_keyframes();

    std::sort(keyfrms.begin(), keyfrms.end(),
              [&](std::shared_ptr<stella_vslam::data::keyframe>& keyfrm_1,
                  std::shared_ptr<stella_vslam::data::keyframe>& keyfrm_2) { return *keyfrm_1 > *keyfrm_2; });

However, this solution is not enough to solve the problem.

## Environment
Hardware: [PC]
CPU: [AMD Ryzen 7 5800X 8-Core Processor]
OS: [Ubuntu 22.04]
In my case it is not necessary to processes the video in real time

ymd-stella, 2022-08-05:
Insufficient information. Please provide complete logs.

ymd-stella, 2022-08-05:
I think the reset occurs because the timestamp is calculated with the start time of the video as 0.
Try disabling the reset.

> You can deactivate the resetting but it does not help to solve the problem.

Insufficient information. Please provide video.

ymd-stella, 2022-08-05:
Issues about resets are handled at https://github.com/stella-cv/stella_vslam/issues/357. Are there any other issues?

ymd-stella, 2022-08-07:
I will close this issue. If you have any additional issues, please create another issue with the missing information.

closed 2022-08-07 by ymd-stella

youknowimcomingwhenyouhearmehumming, 2022-08-07:
Hey :) Thanks for your fast answer.
I have tried both disabling the reset and also setting the appropriate time stamp, but it still fails. I'm away the next few days, but I will get back you Tuesday with an example video and logs of the event. Thank you so much for responding to these issues - it really makes it so much more fun to work with this repository compare to other SLAM repository :)