stella-cv/stella_vslam #370: Unused variables

mitul93, 2022-08-22:
## Describe the bug
Following variables are unused and can be removed.

`frm_obs` in lambda `thread_right`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/system.cc#L326

variable `ini_extractor_left_`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/system.h#L242

variable `bow_db_`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/mapping_module.h#L225

https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/module/initializer.h#L64

variable `bow_vocab_`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/module/loop_detector.h#L126

variable `fix_scale_in_Sim3_estimation_`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/module/loop_detector.h#L137

variable `this`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/stella_vslam/module/local_map_updater.cc#L101

variable `frame_hash_`
https://github.com/stella-cv/stella_vslam/blob/a404f57c1999d8e29475b490e5fe26c38469df53/src/socket_publisher/data_serializer.h#L51

## To Reproduce
The unused variable can be checked by searching them in the vscode or any other editor.

## Expected behavior
Not Applicable

## Screenshots or videos
Not Applicable

## Environment

- Hardware: PC
- CPU: i7-12700
- OS: Ubuntu 22.04
- Commit id: a404f57c1999d8e29475b490e5fe26c38469df53

ymd-stella, 2022-08-22:
LGTM

annavordou, 2022-08-23:
I'm interested in working on this isssue. Could you kindly provide some more information?

ymd-stella, 2022-08-24:
Simply delete unused variables.