stella-cv/stella_vslam #394: OpenMP requires signed integral type as index variables

SteffenEhrle, 2022-09-27:
## Compile error: index Variable in OpenMP 'for' statement must have signed integral type
Using 'unsigned int' in openmp accelerated for-loops leads to compile errors with certain versions of OpenMP. 

## To Reproduce
Compile under Windows using OpenMP (USE_OPENMP):
"error C3016: 'i': index variable in OpenMP 'for' statement must have signed integral type"

## Environment

- Hardware: [PC]
- CPU: [Intel Core i7]
- OS: [Windows 10]
- Commit id: [master]

## Solution
Exchange unsigned int with int64_t in affected for-loops?

ymd-stella, 2022-09-27:
Why can't you use the new version of OpenMP?

SteffenEhrle, 2022-09-27:
I am locked to this version of omp because of other libraries included via vcpkg.

closed 2022-10-08 by ymd-stella