stella-cv/stella_vslam #431: Unable to execute run_video_slam with aist_entrance_hall_2

hofbauerc, 2022-11-23:
Hello,

I am trying to use stella_vslam with ros but I am experiencing errors which nobody seems to have had before. I followed the instructions for the installation based on https://stella-cv.readthedocs.io/en/latest/installation.html. Everything seems to be installed properly and the test as seen in the installation procedure works as intended:
[image]

However when i try to execute a dataset basd on the tutorial: https://stella-cv.readthedocs.io/en/latest/simple_tutorial.html I run into an issue. After reading the config and orb files correctly, the procedure crashes with the following error code:
[image]
The same problem occurs when trying to use stella_vslam_ros. 

Has this issue happened before? What could be the possible reason for this problem to occur?

Thanks in advance.

closed 2022-11-23 by hofbauerc

parsa-yazdankhah, 2023-09-25:
have you found the solution for this specific error? I am having the exact same problem and I haven't been able to solve it for hours.

hofbauerc, 2023-09-25:
No I haven't. We moved on from this approach and didn't need the issue solved. Therefore i closed it. So sorry I cannot help you with it.