stella-cv/stella_vslam #452: CMake Error at cmake/FindSuiteSparse.cmake

Rotoslider, 2023-01-21:
Everything went fine with the install up to the point of:
`cmake \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DINSTALL_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DBUILD_TESTS=ON \
    -DBUILD_EXAMPLES=ON \
    ..`

## Describe the bug

`-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization: DISABLED
-- Use OpenCV 4.2.0
-- aruco: enabled (Found in /usr/include/opencv4/opencv2)
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- gtsam: disabled
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
CMake Error at cmake/FindSuiteSparse.cmake:313 (if):
  if given arguments:

    "EXISTS" "/usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so" "/usr/lib/x86_64-linux-gnu/librt.so"

  Unknown arguments specified
Call Stack (most recent call first):
  src/stella_vslam/CMakeLists.txt:54 (find_package)

## Environment

- Hardware: [PC]
- CPU: [Intel® Celeron(R) N5105 @ 2.00GHz × 4 ]
- OS: [Ubuntu 20.04]
[CMakeOutput.log](https://github.com/stella-cv/stella_vslam/files/10471456/CMakeOutput.log)

[CMakeError.log](https://github.com/stella-cv/stella_vslam/files/10471455/CMakeError.log)

ymd-stella, 2023-01-21:
What is your g2o version or commit hash?

ymd-stella, 2023-01-21:
Updated. You can now use the new g2o, please update stella_vslam to the latest version.

closed 2023-01-21 by ymd-stella