stella-cv/stella_vslam #453: Increase g2o version

ymd-stella, 2023-01-21:
## What issue is the feature request related to?

An older version of g2o is used.

## Describe the solution you'd like

### How to achieve this

- [ ] Update Dockerfile
- [ ] Test/debug

## Additional context

It has been reported that cmake fails when using a newer version of g2o on https://github.com/stella-cv/stella_vslam/issues/452.

closed 2023-01-21 by ymd-stella

billamiable, 2023-03-26:
From the doc, it says the newer g2o uses the latest release. However, when I check g2o repo, I didn't find any releases using this commit. Is there any reason to use this commit? Meanwhile, I see a [newer release being published lately](https://github.com/RainerKuemmerle/g2o/releases/tag/20230223_git), do you intend to upgrade? 

> [g2o](https://github.com/RainerKuemmerle/g2o) : Please use the latest release. Tested on commit ID [ed40a5b](https://github.com/RainerKuemmerle/g2o/tree/ed40a5bb028566fd56a78fd7b04921b613492d6f)

ymd-stella, 2023-04-01:
Updated.

PR #454 (2023-01-21): Update g2o
Close #453