stella-cv/stella_vslam #455: Illegal instruction (core dumped) When trying to run either Simple Tutorial

Rotoslider, 2023-01-21:
## Describe the bug

When running either tutorial from the build folder example:
`./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_1/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg`
I get "Illegal instruction (core dumped)"
Full output:
`(base) lidar@lidar:~/stella_vslam/build$ ./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_3/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg
[2023-01-21 09:40:20.780] [I] config file loaded: ../example/aist/equirectangular.yaml
--start-timestamp is not set. using system timestamp.
If --no-sleep is set without --start-timestamp, timestamps may overlap between multiple runs.
[2023-01-21 09:40:20.781] [I] 
original version of OpenVSLAM,
Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.
stella_vslam (the changes after forking from OpenVSLAM),
Copyright (C) 2022, stella-cv, All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera:
  name: RICOH THETA S 960
  setup: monocular
  model: equirectangular
  fps: 30.0
  cols: 1920
  rows: 960
  color_order: RGB
Preprocessing:
  min_size: 800
  mask_rectangles:
    - [0.0, 1.0, 0.0, 0.1]
    - [0.0, 1.0, 0.84, 1.0]
    - [0.0, 0.2, 0.7, 1.0]
    - [0.8, 1.0, 0.7, 1.0]
Feature:
  name: default ORB feature extraction setting
  scale_factor: 1.2
  num_levels: 8
  ini_fast_threshold: 20
  min_fast_threshold: 7
Mapping:
  backend: g2o
  baseline_dist_thr_ratio: 0.02
  redundant_obs_ratio_thr: 0.95
  num_covisibilities_for_landmark_generation: 20
  num_covisibilities_for_landmark_fusion: 20
Tracking:
  backend: g2o
LoopDetector:
  backend: g2o
  enabled: true
  reject_by_graph_distance: true
  min_distance_on_graph: 50
System:
  map_format: msgpack

[2023-01-21 09:40:20.781] [I] loading ORB vocabulary: ./orb_vocab.fbow
Illegal instruction (core dumped)

I manually download and replaced the lab1 and lab 2 files along with the orb_vocab.flow file in case something was corrupted. I still received the same error afterwards.

## Environment

- Hardware: [PC]
- CPU: [ntel® Celeron(R) N5105 @ 2.00GHz × 4 , Mesa Intel® UHD Graphics (JSL)]
- OS: [e.g. Ubuntu 20.04]

ymd-stella, 2023-01-22:
Show me the result of

```
cat ~/stella_vslam/build/CMakeCache.txt | grep -e USE_ -e BUILD_
```

ymd-stella, 2023-01-22:
Add the following to the build options for stella_vslam.

```
-DUSE_STACK_TRACE_LOGGER=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -USE_SSE_ORB=OFF
```

Rotoslider, 2023-01-22:
> 

`BUILD_EXAMPLES:BOOL=ON
BUILD_GMOCK:BOOL=ON
BUILD_SHARED_LIBS:BOOL=ON
BUILD_TESTS:BOOL=ON
BUILD_UTILS:BOOL=OFF
BUILD_WITH_MARCH_NATIVE:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=
USE_ARUCO:BOOL=ON
USE_AVX:BOOL=ON
USE_CCACHE:BOOL=ON
USE_CONTRIB:BOOL=OFF
USE_FAST_MATH:BOOL=OFF
USE_GOOGLE_PERFTOOLS:BOOL=OFF
USE_GTSAM:BOOL=ON
USE_MMX:BOOL=ON
USE_O3:BOOL=ON
USE_OPENMP:BOOL=OFF
USE_OWN_EIGEN3:BOOL=ON
USE_PANGOLIN_VIEWER:BOOL=ON
USE_SANITIZER:BOOL=OFF
USE_SOCKET_PUBLISHER:BOOL=OFF
USE_SSE:BOOL=ON
USE_SSE2:BOOL=ON
USE_SSE3:BOOL=ON
USE_SSE4:BOOL=ON
USE_SSE_FP_MATH:BOOL=OFF
USE_SSE_ORB:BOOL=OFF
USE_STACK_TRACE_LOGGER:BOOL=OFF
//STRINGS property for variable: CMAKE_BUILD_TYPE
CMAKE_BUILD_TYPE-STRINGS:INTERNAL=Debug;Release

`

Rotoslider, 2023-01-22:
I deleted the stella folder and re downloaded the git. Now when I try to build I get a new error.
`CMake Error at example/CMakeLists.txt:11 (find_package):
  By not providing "FindBackward.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Backward",
  but CMake did not find one.

  Could not find a package configuration file provided by "Backward" with any
  of the following names:

    BackwardConfig.cmake
    backward-config.cmake`

Full error here:
`CMake Error at example/CMakeLists.txt:11 (find_package):
  By not providing "FindBackward.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Backward",
  but CMake did not find one.

  Could not find a package configuration file provided by "Backward" with any
  of the following names:

    BackwardConfig.cmake
    backward-config.cmake`
[CMakeOutput.log](https://github.com/stella-cv/stella_vslam/files/10475406/CMakeOutput.log)

[CMakeError.log](https://github.com/stella-cv/stella_vslam/files/10475405/CMakeError.log)

ymd-stella, 2023-01-23:
Install backward-cpp with the following command.

```bash
git clone https://github.com/bombela/backward-cpp.git
cd backward-cpp
git checkout 5ffb2c879ebdbea3bdb8477c671e32b1c984beaa
mkdir -p build
cd build
cmake \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    ..
make -j
make install
```

Rotoslider, 2023-01-23:
That fixed Backwards but now I have new errors.
` Building CXX object src/stella_vslam/CMakeFiles/stella_vslam.dir/io/map_database_io_sqlite3.cc.o
/home/lidar/stella_vslam/src/stella_vslam/io/map_database_io_sqlite3.cc: In member function ‘virtual void stella_vslam::io::map_database_io_sqlite3::save(const string&, const stella_vslam::data::camera_database*, const stella_vslam::data::orb_params_database*, const stella_vslam::data::map_database*)’:
/home/lidar/stella_vslam/src/stella_vslam/io/map_database_io_sqlite3.cc:20:75: warning: unused parameter ‘orb_params_db’ [-Wunused-parameter]
   20 |                                    const data::orb_params_database* const orb_params_db,
`

` Building CXX object src/stella_vslam/CMakeFiles/stella_vslam.dir/match/fuse.cc.o
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc: In function ‘cv::Ptr<cv::aruco::Dictionary> {anonymous}::get_predefined_dictionary(int, int)’:
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:12:16: error: ‘PREDEFINED_DICTIONARY_NAME’ is not a member of ‘cv::aruco’
   12 |     cv::aruco::PREDEFINED_DICTIONARY_NAME dictionary_name = cv::aruco::DICT_4X4_50;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:14:9: error: ‘dictionary_name’ was not declared in this scope
   14 |         dictionary_name = cv::aruco::DICT_4X4_50;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:17:9: error: ‘dictionary_name’ was not declared in this scope
   17 |         dictionary_name = cv::aruco::DICT_4X4_100;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:20:9: error: ‘dictionary_name’ was not declared in this scope
   20 |         dictionary_name = cv::aruco::DICT_4X4_250;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:23:9: error: ‘dictionary_name’ was not declared in this scope
   23 |         dictionary_name = cv::aruco::DICT_4X4_1000;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:26:9: error: ‘dictionary_name’ was not declared in this scope
   26 |         dictionary_name = cv::aruco::DICT_5X5_50;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:29:9: error: ‘dictionary_name’ was not declared in this scope
   29 |         dictionary_name = cv::aruco::DICT_5X5_100;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:32:9: error: ‘dictionary_name’ was not declared in this scope
   32 |         dictionary_name = cv::aruco::DICT_5X5_250;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:35:9: error: ‘dictionary_name’ was not declared in this scope
   35 |         dictionary_name = cv::aruco::DICT_5X5_1000;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:38:9: error: ‘dictionary_name’ was not declared in this scope
   38 |         dictionary_name = cv::aruco::DICT_6X6_50;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:41:9: error: ‘dictionary_name’ was not declared in this scope
   41 |         dictionary_name = cv::aruco::DICT_6X6_100;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:44:9: error: ‘dictionary_name’ was not declared in this scope
   44 |         dictionary_name = cv::aruco::DICT_6X6_250;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:47:9: error: ‘dictionary_name’ was not declared in this scope
   47 |         dictionary_name = cv::aruco::DICT_6X6_1000;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:50:9: error: ‘dictionary_name’ was not declared in this scope
   50 |         dictionary_name = cv::aruco::DICT_7X7_50;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:53:9: error: ‘dictionary_name’ was not declared in this scope
   53 |         dictionary_name = cv::aruco::DICT_7X7_100;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:56:9: error: ‘dictionary_name’ was not declared in this scope
   56 |         dictionary_name = cv::aruco::DICT_7X7_250;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:59:9: error: ‘dictionary_name’ was not declared in this scope
   59 |         dictionary_name = cv::aruco::DICT_7X7_1000;
      |         ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:64:47: error: ‘dictionary_name’ was not declared in this scope
   64 |     return cv::aruco::getPredefinedDictionary(dictionary_name);
      |                                               ^~~~~~~~~~~~~~~
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc: In constructor ‘stella_vslam::marker_detector::aruco::aruco(const stella_vslam::camera::base*, const std::shared_ptr<stella_vslam::marker_model::base>&, unsigned int, double)’:
/home/lidar/stella_vslam/src/stella_vslam/marker_detector/aruco.cc:75:50: error: ‘create’ is not a member of ‘cv::aruco::DetectorParameters’
   75 |     parameters_ = cv::aruco::DetectorParameters::create();
[... cut ...]
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR) 
-- Found libbfd: /usr/lib/x86_64-linux-gnu/libbfd.so  
-- Found libdwarf: /usr/lib/x86_64-linux-gnu/libdwarf.so  
-- Found Backward: /usr/local/lib/backward  
-- Stack trace logger: ENABLED
-- Google Perftools: DISABLED
-- Viewer for examples: PangolinViewer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lidar/stella_vslam/build
`
Opencv 4.7 with aruco installed. But this may be the cause. When it was 4.6 this last error was not present.

Rotoslider, 2023-01-24:
I uninstalled opencv4.7 and tried to build 4.6. It however now will not build in my system. So I rebuilt 4.7 without issue. I will wait for your enhancement of being able to use opencv4.7

ymd-stella, 2023-01-28:
Updated. https://github.com/stella-cv/stella_vslam/pull/460

Rotoslider, 2023-01-30:
Created a new stella_vslam folder and cloned the latest version. 
Seemed everything built fine and installed. However I get a new error when trying to run:
`./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_1/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg
`
After the yaml loads i get:
`2023-01-29 18:58:09.799] [I] loading ORB vocabulary: ./orb_vocab.fbow
Stack trace (most recent call last):
#6    Object "", at 0xffffffffffffffff, in 
#5    Object "/home/lidar/stella_vslam/build/run_video_slam", at 0x55eba56866fd, in _start
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2e8238d082, in __libc_start_main
#3  | Source "/home/lidar/stella_vslam/example/run_video_slam.cc", line 258, in main
    |   257:     // build a slam system
    | > 258:     auto slam = std::make_shared<stella_vslam::system>(cfg, vocab_file_path->value());
    |   259:     bool need_initialize = true;
    |   260:     if (map_db_path_in->is_set()) {
    | Source "/usr/include/c++/9/bits/shared_ptr.h", line 718, in std::shared_ptr<stella_vslam::system> std::make_shared<stella_vslam::system, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |   716:       typedef typename std::remove_cv<_Tp>::type _Tp_nc;
    |   717:       return std::allocate_shared<_Tp>(std::allocator<_Tp_nc>(),
    | > 718: 				       std::forward<_Args>(__args)...);
    |   719:     }
    | Source "/usr/include/c++/9/bits/shared_ptr.h", line 702, in std::shared_ptr<stella_vslam::system> std::allocate_shared<stella_vslam::system, std::allocator<stella_vslam::system>, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::allocator<stella_vslam::system> const&, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |   700:     {
    |   701:       return shared_ptr<_Tp>(_Sp_alloc_shared_tag<_Alloc>{__a},
    | > 702: 			     std::forward<_Args>(__args)...);
    |   703:     }
    | Source "/usr/include/c++/9/bits/shared_ptr.h", line 359, in std::shared_ptr<stella_vslam::system>::shared_ptr<std::allocator<stella_vslam::system>, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<stella_vslam::system> >, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |   357:       template<typename _Alloc, typename... _Args>
    |   358: 	shared_ptr(_Sp_alloc_shared_tag<_Alloc> __tag, _Args&&... __args)
    | > 359: 	: __shared_ptr<_Tp>(__tag, std::forward<_Args>(__args)...)
    |   360: 	{ }
    | Source "/usr/include/c++/9/bits/shared_ptr_base.h", line 1344, in std::__shared_ptr<stella_vslam::system, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<stella_vslam::system>, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::_Sp_alloc_shared_tag<std::allocator<stella_vslam::system> >, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |  1342:       template<typename _Alloc, typename... _Args>
    |  1343: 	__shared_ptr(_Sp_alloc_shared_tag<_Alloc> __tag, _Args&&... __args)
    | >1344: 	: _M_ptr(), _M_refcount(_M_ptr, __tag, std::forward<_Args>(__args)...)
    |  1345: 	{ _M_enable_shared_from_this_with(_M_ptr); }
    | Source "/usr/include/c++/9/bits/shared_ptr_base.h", line 679, in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<stella_vslam::system, std::allocator<stella_vslam::system>, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(stella_vslam::system*&, std::_Sp_alloc_shared_tag<std::allocator<stella_vslam::system> >, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |   677: 	  auto __guard = std::__allocate_guarded(__a2);
    |   678: 	  _Sp_cp_type* __mem = __guard.get();
    | > 679: 	  auto __pi = ::new (__mem)
    |   680: 	    _Sp_cp_type(__a._M_a, std::forward<_Args>(__args)...);
    |   681: 	  __guard = nullptr;
    | Source "/usr/include/c++/9/bits/shared_ptr_base.h", line 548, in std::_Sp_counted_ptr_inplace<stella_vslam::system, std::allocator<stella_vslam::system>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::allocator<stella_vslam::system>, std::shared_ptr<stella_vslam::config>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)
    |   546: 	  // _GLIBCXX_RESOLVE_LIB_DEFECTS
    |   547: 	  // 2070.  allocate_shared should use allocator_traits<A>::construct
    | > 548: 	  allocator_traits<_Alloc>::construct(__a, _M_ptr(),
    |   549: 	      std::forward<_Args>(__args)...); // might throw
    |   550: 	}
[... cut ...]
Illegal instruction (core dumped)`
Results of:
`cat ~/stella_vslam/build/CMakeCache.txt | grep -e USE_ -e BUILD_`
`BUILD_EXAMPLES:BOOL=ON
BUILD_SHARED_LIBS:BOOL=ON
BUILD_TESTS:BOOL=OFF
BUILD_WITH_MARCH_NATIVE:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
USE_ARUCO:BOOL=ON
USE_CCACHE:BOOL=ON
USE_GOOGLE_PERFTOOLS:BOOL=OFF
USE_GTSAM:BOOL=ON
USE_OPENMP:BOOL=OFF
USE_PANGOLIN_VIEWER:BOOL=ON
USE_SANITIZER:BOOL=OFF
USE_SOCKET_PUBLISHER:BOOL=OFF
USE_SSE_FP_MATH:BOOL=OFF
USE_SSE_ORB:BOOL=OFF
USE_STACK_TRACE_LOGGER:BOOL=ON`

ymd-stella, 2023-01-30:
Rebuild FBoW with RelWithDebInfo.

```bash
cd /tmp
git clone https://github.com/stella-cv/FBoW.git
cd FBoW
mkdir build && cd build
cmake \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DCMAKE_INSTALL_PREFIX=/usr/local \
    ..
make -j4
sudo make install
```

Rotoslider, 2023-01-31:
Closer. The video came up and pangolin_viewer started then both winked out in a couple seconds. Followed by the error below.
ran:
`./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_1/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg`

`2023-01-30 17:32:52.103] [I] loading ORB vocabulary: ./orb_vocab.fbow
[2023-01-30 17:32:52.185] [I] load orb_params "default ORB feature extraction setting"
[2023-01-30 17:32:52.187] [I] startup SLAM system
[2023-01-30 17:32:52.187] [I] start mapping module
[2023-01-30 17:32:52.187] [I] start global optimization module
[2023-01-30 17:32:52.436] [I] initialization succeeded with E
Stack trace (most recent call last) in thread 4658:
#11   Object "", at 0xffffffffffffffff, in 
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)
#10   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f407a5e2132, in clone
#9    Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f407bd3a608]
#8    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f407a7a6de3, in std::error_code::default_error_condition() const
#7  | Source "/usr/include/c++/9/thread", line 195, in _M_run
    |   194: 	void
    | > 195: 	_M_run() { _M_func(); }
    |   196:       };
    | Source "/usr/include/c++/9/thread", line 251, in operator()
    |   249: 	  using _Indices
    |   250: 	    = typename _Build_index_tuple<tuple_size<_Tuple>::value>::__type;
    | > 251: 	  return _M_invoke(_Indices());
    |   252: 	}
    |   253:       };
    | Source "/usr/include/c++/9/thread", line 244, in _M_invoke<0>
    |   242: 	  typename __result<_Tuple>::type
    |   243: 	  _M_invoke(_Index_tuple<_Ind...>)
    | > 244: 	  { return std::__invoke(std::get<_Ind>(std::move(_M_t))...); }
    |   245: 
    |   246: 	typename __result<_Tuple>::type
    | Source "/usr/include/c++/9/bits/invoke.h", line 95, in __invoke<mono_tracking(const std::shared_ptr<stella_vslam::system>&, const std::shared_ptr<stella_vslam::config>&, const string&, const string&, unsigned int, unsigned int, bool, bool, bool, const string&, const string&, double)::<lambda()> >
    |    93:       using __type = typename __result::type;
    |    94:       using __tag = typename __result::__invoke_type;
    | >  95:       return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
    |    96: 					std::forward<_Args>(__args)...);
    |    97:     }
    | Source "/usr/include/c++/9/bits/invoke.h", line 60, in __invoke_impl<void, mono_tracking(const std::shared_ptr<stella_vslam::system>&, const std::shared_ptr<stella_vslam::config>&, const string&, const string&, unsigned int, unsigned int, bool, bool, bool, const string&, const string&, double)::<lambda()> >
    |    58:     constexpr _Res
    |    59:     __invoke_impl(__invoke_other, _Fn&& __f, _Args&&... __args)
    | >  60:     { return std::forward<_Fn>(__f)(std::forward<_Args>(__args)...); }
    |    61: 
    |    62:   template<typename _Res, typename _MemFun, typename _Tp, typename... _Args>
      Source "/home/lidar/stella_vslam/example/run_video_slam.cc", line 90, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<mono_tracking(std::shared_ptr<stella_vslam::system> const&, std::shared_ptr<stella_vslam::config> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)::{lambda()#1}> > >::_M_run() [0x55d63321abce]
         88:             if (!frame.empty() && (num_frame % frame_skip == 0)) {
         89:                 // input the current frame and estimate the camera pose
      >  90:                 slam->feed_monocular_frame(frame, timestamp, mask);
         91:             }
         92: 
         93:             const auto tp_2 = std::chrono::steady_clock::now();
#6    Source "/home/lidar/stella_vslam/src/stella_vslam/system.cc", line 439, in stella_vslam::system::feed_monocular_frame(cv::Mat const&, double, cv::Mat const&) [0x7f407be853ee]
        436:         spdlog::warn("preprocess: empty image");
        437:         return nullptr;
        438:     }
      > 439:     return feed_frame(create_monocular_frame(img, timestamp, mask), img);
        440: }
        441: 
        442: std::shared_ptr<Mat44_t> system::feed_stereo_frame(const cv::Mat& left_img, const cv::Mat& right_img, const double timestamp, const cv::Mat& mask) {
#5    Source "/home/lidar/stella_vslam/src/stella_vslam/system.cc", line 465, in stella_vslam::system::feed_frame(stella_vslam::data::frame const&, cv::Mat const&) [0x7f407be82732]
        463:     const auto start = std::chrono::system_clock::now();
        464: 
      > 465:     const auto cam_pose_wc = tracker_->feed_frame(frm);
        466: 
        467:     const auto end = std::chrono::system_clock::now();
        468:     double elapsed_ms = std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
#4    Source "/home/lidar/stella_vslam/src/stella_vslam/tracking_module.cc", line 125, in stella_vslam::tracking_module::feed_frame(stella_vslam::data::frame) [0x7f407be9b0c7]
        123:     bool succeeded = false;
[... cut ...]
    |    63:     inline bool isSafeAVX(){return  HW_AVX && OS_AVX;}
      Source "/usr/include/x86_64-linux-gnu/bits/string_fortified.h", line 71, in fbow::Vocabulary::transform(cv::Mat const&, int, fbow::BoWVector&, fbow::BoWFeatVector&) [0x7f407a12f188]
         68:       return __dest;
         69:     }
         70: #endif
      >  71:   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
         72: }
         73: 
         74: #ifdef __USE_MISC
Illegal instruction (Illegal operand [0x7f407a12f188])
Illegal instruction (core dumped)`

Rotoslider, 2023-02-04:
This works now. Thank you very much for taking the time to help.

closed 2023-02-04 by Rotoslider