stella-cv/stella_vslam #457: fatal error: tinycolormap.hpp: No such file or directory

zoldaten, 2023-01-24:
## Describe the bug
following the instruction - https://stella-cv.readthedocs.io/en/latest/installation.html#subsection-common-linux-macos
cant build stella-vslam

## To Reproduce
When running `make -j4` I get 
```
[ 61%] Building CXX object src/stella_vslam/CMakeFiles/stella_vslam.dir/optimize/terminate_action.cc.o
[ 62%] Building CXX object src/stella_vslam/CMakeFiles/stella_vslam.dir/publish/frame_publisher.cc.o
/home/al/stella_vslam/src/stella_vslam/publish/frame_publisher.cc:9:10: fatal error: tinycolormap.hpp: No such file or directory
    9 | #include <tinycolormap.hpp>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/stella_vslam/CMakeFiles/stella_vslam.dir/build.make:817: src/stella_vslam/CMakeFiles/stella_vslam.dir/publish/frame_publisher.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:516: src/stella_vslam/CMakeFiles/stella_vslam.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

```
build options:
cmake \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DINSTALL_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DBUILD_TESTS=OFF \
    -DBUILD_EXAMPLES=ON \
    ..
```

```
cat CMakeCache.txt | grep -e USE_ -e BUILD_

BUILD_EXAMPLES:BOOL=ON
BUILD_SHARED_LIBS:BOOL=ON
BUILD_TESTS:BOOL=OFF
BUILD_WITH_MARCH_NATIVE:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
USE_ARUCO:BOOL=ON
USE_CCACHE:BOOL=ON
USE_GOOGLE_PERFTOOLS:BOOL=OFF
USE_GTSAM:BOOL=ON
USE_OPENMP:BOOL=OFF
USE_PANGOLIN_VIEWER:BOOL=ON
USE_SANITIZER:BOOL=OFF
USE_SOCKET_PUBLISHER:BOOL=OFF
USE_SSE_FP_MATH:BOOL=OFF
USE_SSE_ORB:BOOL=OFF
USE_STACK_TRACE_LOGGER:BOOL=ON
```

## Environment
- Hardware: mini PC
- CPU: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- OS: Ubuntu 20.04.5 LTS. 
`Linux al-i7 5.15.0-58-generic #64~20.04.1-Ubuntu SMP Fri Jan 6 16:42:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`
- Install procedure: native

ymd-stella, 2023-01-24:
submodule has not been downloaded.
See https://stella-cv.readthedocs.io/en/latest/installation.html#source-code

ymd-stella, 2023-01-24:
If you are sure that tinycolormap.hpp exists in stella_vslam/3rd/tinycolormap, describe the log and reopen this.

closed 2023-01-24 by ymd-stella

zoldaten, 2023-01-24:
sorry, you are right. silly mistake.