stella-cv/stella_vslam #479: Docker Error - CMake 3.14 or higher is required.

ajsnyde, 2023-02-22:
## Describe the bug

Docker CMake version is too old to compile g2o.

`CMake 3.14 or higher is required.  You are running version 3.10.2`

## To Reproduce

Running `sudo docker build -t stella_vslam-desktop -f Dockerfile.desktop . --build-arg NUM_THREADS=4`, and get this:

```ERROR: failed to solve: process "/bin/sh -c set -x &&   git clone https://github.com/RainerKuemmerle/g2o.git &&   cd g2o &&   git checkout ${G2O_COMMIT} &&   mkdir -p build &&   cd build &&   cmake     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}     -DBUILD_SHARED_LIBS=ON     -DBUILD_UNITTESTS=OFF     -DG2O_USE_CHOLMOD=OFF     -DG2O_USE_CSPARSE=ON     -DG2O_USE_OPENGL=OFF     -DG2O_USE_OPENMP=OFF     -DG2O_BUILD_APPS=OFF     -DG2O_BUILD_EXAMPLES=OFF     -DG2O_BUILD_LINKED_APPS=OFF     .. &&   make -j${NUM_THREADS} &&   make install &&   cd /tmp &&   rm -rf *" did not complete successfully: exit code: 1
~/stella_vslam (main)$ sudo docker build -t stella_vslam-desktop -f Dockerfile.desktop . --build-arg NUM_THREADS=13
[+] Building 3.9s (10/20)                                                                                                                                                           
 => [internal] load .dockerignore                                                                                                                                              0.0s
 => => transferring context: 308B                                                                                                                                              0.0s
 => [internal] load build definition from Dockerfile.desktop                                                                                                                   0.0s
 => => transferring dockerfile: 6.01kB                                                                                                                                         0.0s
 => [internal] load metadata for docker.io/library/ubuntu:18.04                                                                                                                0.3s
 => [internal] load build context                                                                                                                                              0.0s
 => => transferring context: 27.88kB                                                                                                                                           0.0s
 => [ 1/16] FROM docker.io/library/ubuntu:18.04@sha256:a3765b4d74747b5e9bdd03205b3fbc4fa19a02781c185f97f24c8f4f84ed7bbf                                                        0.0s
 => CACHED [ 2/16] RUN set -x &&   apt-get update -y -qq &&   apt-get upgrade -y -qq --no-install-recommends &&   : "basic dependencies" &&   apt-get install -y -qq     buil  0.0s
 => CACHED [ 3/16] WORKDIR /tmp                                                                                                                                                0.0s
 => CACHED [ 4/16] RUN set -x &&   wget -q https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.tar.bz2 &&   tar xf eigen-3.3.7.tar.bz2 &&   rm -rf eigen-3.3.7.tar.  0.0s
 => CACHED [ 5/16] WORKDIR /tmp                                                                                                                                                0.0s
 => ERROR [ 6/16] RUN set -x &&   git clone https://github.com/RainerKuemmerle/g2o.git &&   cd g2o &&   git checkout ed40a5bb028566fd56a78fd7b04921b613492d6f &&   mkdir -p b  3.5s
------                                                                                    ```

shell session:

```shell-session
foo@~/lib/stella_vslam/build$ command
some output ...
"<error>"
```

build options:

```
foo@~/lib/stella_vslam/build: cat CMakeCache.txt | grep -e USE_ -e BUILD_
BUILD_EXAMPLES:BOOL=ON
BUILD_SHARED_LIBS:BOOL=ON
BUILD_TESTS:BOOL=OFF
BUILD_WITH_MARCH_NATIVE:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
USE_ARUCO:BOOL=ON
USE_CCACHE:BOOL=ON
USE_GOOGLE_PERFTOOLS:BOOL=OFF
USE_GTSAM:BOOL=ON
USE_OPENMP:BOOL=OFF
USE_PANGOLIN_VIEWER:BOOL=OFF
USE_SANITIZER:BOOL=OFF
USE_SOCKET_PUBLISHER:BOOL=ON
USE_SSE_FP_MATH:BOOL=OFF
USE_SSE_ORB:BOOL=OFF
USE_STACK_TRACE_LOGGER:BOOL=ON
```

ajsnyde, 2023-02-22:
I corrected the issue by changing the Docker file to use Ubuntu 20.04 instead of 18.04. I haven't tested any functionality yet, I was just about to build the image.

ajsnyde, 2023-02-22:
Oh lol you just did that. Nice! Is there a way I can donate a little? I appreciate the super-fast response time

ymd-stella, 2023-02-22:
I think it will work on 20.04 as it has been tested by CI.

ymd-stella, 2023-02-22:
Just reporting bugs would be appreciated.

PR #480 (2023-02-22): Deprecated ubuntu 18.04
Close #479
closed 2023-02-22 by ymd-stella