stella-cv/stella_vslam #481: Map Viewer Window  not showing anything

ajsnyde, 2023-02-22:
Not sure what could be causing this. Running the basic tutorial after successfully building the docker. Host machine is Ubuntu 22.04 with xfce installed on top.
https://user-images.githubusercontent.com/11302823/220725247-fdf29955-0595-46f3-8a39-a3d24d41aa92.mp4

ymd-stella, 2023-02-23:
- Is a nvidia GPU installed?
- Does the same thing happen if you run `export LIBGL_ALWAYS_SOFTWARE=1` before running stella_vslam?
- Does the same thing happen if you run docker with `--cap-add IPC_OWNER`?
- Does the same thing happen if you run docker with `--ipc host`?

ajsnyde, 2023-02-23:
-     Is a nvidia GPU installed?
    - yes, a 2080
-     Does glxgears work properly?
    - yes, on the host machine it runs fine
I added those arguments like so, without it fixing the issue:
` sudo docker run -it --rm --cap-add IPC_OWNER --ipc host --privileged -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix:ro stella_vslam-desktop`

Finally, I ensured that `export LIBGL_ALWAYS_SOFTWARE=1` was run before running the basic tutorial command:
`./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_1/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg`, of course after downloading the vocab and materials listed in the tutorial.

As a workaround, I'm using the server/socket to run and display right now.

ymd-stella, 2023-02-23:
Do not use sudo to do `docker run`. See https://docs.docker.com/engine/install/linux-postinstall/

ymd-stella, 2023-02-23:
> yes, on the host machine it runs fine

What about when run inside a container?

Jlim01, 2023-03-09:
Vocabulary::readFromFile could not open:./orb_vocab.fbow

Jlim01, 2023-03-09:
Can I visualize without gpu? I have just Cpu

ymd-stella, 2023-03-09:
> Can I visualize without gpu? I have just Cpu

It worked without GPU on Ubuntu 18.04. I think it will work on Ubuntu 20.04 or later, but it doesn't seem to work in some environments.

ymd-stella, 2023-03-21:
In my environment, I solved the problem by adding `--privileged`. [documentation](https://stella-cv.readthedocs.io/en/latest/docker.html) has been updated.