stella-cv/stella_vslam #487: can't open /orb _vocab.fbow

Jlim01, 2023-03-09:
I use Intel cpu not gpu (I don't have gpu like nvidia)
My ubuntu environment is that 18.04 ver (open cv 3.x)
When i input run command on the docker environment, It happened under the error. 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Vocabulary::readFromFile could not open:./orb_vocab.fbow
How to solve this error? Im just wanna see VSLAM(like ORB2) visualizing

`root@88554c449440:/stella_vslam/build# ./run_video_slam --disable-mapping -v ./orb_vocab.fbow -m ./aist_living_lab_2/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-in map.msg
[2023-03-09 03:00:35.911] [I] config file loaded: ../example/aist/equirectangular.yaml
--start-timestamp is not set. using system timestamp.
If --no-sleep is set without --start-timestamp, timestamps may overlap between multiple runs.
[2023-03-09 03:00:35.911] [I] 
original version of OpenVSLAM,
Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.
stella_vslam (the changes after forking from OpenVSLAM),
Copyright (C) 2022, stella-cv, All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera:
  name: RICOH THETA S 960
  setup: monocular
  model: equirectangular
  fps: 30.0
  cols: 1920
  rows: 960
  color_order: RGB
Preprocessing:
  min_size: 800
  mask_rectangles:
    - [0.0, 1.0, 0.0, 0.1]
    - [0.0, 1.0, 0.84, 1.0]
    - [0.0, 0.2, 0.7, 1.0]
    - [0.8, 1.0, 0.7, 1.0]
Feature:
  name: default ORB feature extraction setting
  scale_factor: 1.2
  num_levels: 8
  ini_fast_threshold: 20
  min_fast_threshold: 7
Mapping:
  backend: g2o
  baseline_dist_thr_ratio: 0.02
  redundant_obs_ratio_thr: 0.95
  num_covisibilities_for_landmark_generation: 20
  num_covisibilities_for_landmark_fusion: 20
Tracking:
  backend: g2o
LoopDetector:
  backend: g2o
  enabled: true
  reject_by_graph_distance: true
  min_distance_on_graph: 50
System:
  map_format: msgpack

[2023-03-09 03:00:35.911] [I] loading ORB vocabulary: ./orb_vocab.fbow
terminate called after throwing an instance of 'std::runtime_error'
  what():  Vocabulary::readFromFile could not open:./orb_vocab.fbow
Stack trace (most recent call last):
#14   Object "", at 0xffffffffffffffff, in 
#13   Object "/stella_vslam/build/run_video_slam", at 0x564886f1ce44, in _start
#12   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff078bc5e3f, in __libc_start_main
#11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff078bc5d8f, in __libc_init_first
#10 | Source "/stella_vslam/example/run_video_slam.cc", line 267, in main
    |   266:     // build a slam system
[... 55 lines cut ...]
         36:         exit(EXIT_FAILURE);
         37:     }
         38: #else
      >  39:     bow_vocab->readFromFile(path);
         40:     if (!bow_vocab->isValid()) {
         41:         spdlog::critical("wrong path to vocabulary");
         42:         delete bow_vocab;
#7    Source "/stella_vslam/3rd/FBoW/src/fbow.cpp", line 391, in fbow::Vocabulary::readFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [0x7ff07884360d]
        388: //loads/saves from a file
        389: void Vocabulary::readFromFile(const std::string &filepath){
        390:     std::ifstream file(filepath,std::ios::binary);
      > 391:     if (!file) throw std::runtime_error("Vocabulary::readFromFile could not open:"+filepath);
        392:     fromStream(file);
        393: }
#6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff078e92517, in __cxa_throw
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff078e922b6, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff078e9224b, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7ff078e86bbd, in __cxa_throw_bad_array_new_length
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff078bc47f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff078bde475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff078c32a7c, in pthread_kill
Aborted (Signal sent by tkill() 44 0)
Aborted (core dumped)
root@88554c449440:/stella_vslam/build# 
`

Jlim01, 2023-03-09:
The problem was that the files could not be downloaded. Re-downloading to docker solved the problem. However, as in the question below, the pangolin viewer (map view) does not appear as a window in Docker. Of course, the cloud point viewer comes out as a window.

[image]

Jlim01, 2023-03-09:
solved problem. but there is other problem (map viewer)

Jlim01, 2023-03-09:
[image]

my command 
./run_video_slam -v ./orb_vocab.fbow -m ./aist_living_lab_1/video.mp4 -c ../example/aist/equirectangular.yaml --frame-skip 3 --no-sleep --map-db-out map.msg

ymd-stella, 2023-03-09:
Close because it is duplicated with #481.

closed 2023-03-09 by ymd-stella