stella-cv/stella_vslam #488: Cannnot run Stereo SLAM -> Invalid setup type: Stereo

Shahrullo, 2023-03-16:
When running stereo configuration like:

./run_video_slam -v ../vocab/orb_vocab.fbow -m ../dataset/slam_2023-03-15_15-12-52.mp4 -c ../example/custom/stereo.yaml --frame-skip 3 --no-sleep --map-db-out map_stereo.msg

I'm getting the error: `Invalid setup type: Stereo`

` throw std::runtime_error("Invalid setup type: " + slam->get_camera()->get_setup_type_string());`

Here is my `.yaml` configuration: 

```
# EuRoC stereo model

# intrinsic parameters after stereo-rectification are written
Camera:
  name: "ELP-CMOS-OV9750"
  setup: "stereo"
  model: "perspective"

# new "rectified" matrices is the first three cols of the projection matrix which is calculated with cv::stereoRectify()
# e.g. fx = P1[0][0] or P2[0][0], cx = P1[0][2] or P2[0][2]
#      fy = P1[1][1] or P2[1][1], cy = P1[1][2] or P2[1][2]

  fx: 1150.064453125
  fy: 1150.064453125
  cx: 484.0756149291992
  cy: 491.6306686401367

# there is no distortion after stereo rectification

  k1: 0.0
  k2: 0.0
  p1: 0.0
  p2: 0.0
  k3: 0.0

# focal_x_baseline is -P2[0][3] which is calculated with cv::stereoRectify()

  fps: 30
  cols: 2560
  rows: 720
  focal_x_baseline: 54767.74343744639

  color_order: "Gray"

# original intrinsic parameters (K, D) and stereo-recitification parameters (R)
# matrices (K, R) are written in row-major order

StereoRectifier:
  K_left: [1.15860083e+03, 0.0, 6.94914541e+02, 0.0, 1.16073792e+03, 4.91680888e+02, 0.0, 0.0, 1.0]
  D_left: [0.0957377, -0.20468165, -0.00156101, -0.00057928, 0.12717208]
  R_left: [0.98961395, -0.00507419, 0.14366097, 0.00527846, 0.99998553, -0.00104082, -0.1436536, 0.00178832, 0.98962642]
  K_right: [1.14149890e+03, 0.0, 6.89591244e+02, 0.0, 1.13939099e+03, 4.93099586e+02, 0.0, 0.0, 1.0]
  D_right: [0.06340865, -0.00211256, -0.00047321, -0.0009633, -0.28608742]
  R_right: [0.991024, -0.00588789, 0.13355437, 0.00569804, 0.99998214, 0.00180366, -0.335626, -0.00102648, 0.99103985]

Preprocessing:
  min_size: 800

Feature:
  name: "default ORB feature extraction setting"
  scale_factor: 1.2
  num_levels: 8
  ini_fast_threshold: 20
  min_fast_threshold: 7

Mapping:
  backend: "g2o"
  baseline_dist_thr: 0.11007784219
  redundant_obs_ratio_thr: 0.9

Tracking:
  backend: "g2o"

LoopDetector:
  backend: "g2o"

Initializer:
  num_min_triangulated_pts: 100

System:
  map_format: "msgpack"

PangolinViewer:
  keyframe_size: 0.07
  keyframe_line_width: 1
  graph_line_width: 1
  point_size: 2
  camera_size: 0.08
  camera_line_width: 3
  viewpoint_x: 0
  viewpoint_y: -0.65
  viewpoint_z: -1.9
  viewpoint_f: 400

```

ymd-stella, 2023-03-16:
Your configuration file is incorrect.

closed 2023-03-16 by ymd-stella

ymd-stella, 2023-03-16:
This does not qualify as a bug report. Please use the template. The next time you do something similar, we may block you. Please be careful not to do so.

ymd-stella, 2023-03-16:
Maybe you should post your question in Discussions first.