stella-cv/stella_vslam #513: Compilation error: ‘number_t’ does not name a type

Nam-Nguyen-Hoang, 2023-06-30:
## Describe the bug
On installing this model, I got the bug as written in the title: ‘number_t’ does not name a type. Is it a custom type?

## To Reproduce

Prerequisite installation:
System: Ubuntu 22.04
```
sudo apt-get install -y libeigen3-dev libmetis-dev libbfd-dev libdw-dev libdwarf-dev libelf-dev libspdlog-dev

git clone https://github.com/stella-cv/FBoW.git
cd FBoW && mkdir build && cd build
cmake .. -DBUILD_TESTS=ON -DBUILD_UTILS=ON
sudo make install
cd ../..

git clone https://github.com/bombela/backward-cpp.git
cd backward-cpp && mkdir build && cd build
cmake ..
sudo make install
cd ../..

git clone --recursive https://github.com/stevenlovegrove/Pangolin.git
cd Pangolin
./scripts/install_prerequisites.sh recommended
cmake -B build -GNinja
cmake --build build -t pypangolin_pip_install
cd ..
```

build options:

```
git clone https://github.com/stella-cv/stella_vslam.git
cd stella_vslam && mkdir build && cd build
cmake \
    -DUSE_STACK_TRACE_LOGGER=ON \
    -DCMAKE_BUILD_TYPE=RelWithDebInfo \
    -DUSE_PANGOLIN_VIEWER=ON \
    -DINSTALL_PANGOLIN_VIEWER=ON \
    -DUSE_SOCKET_PUBLISHER=OFF \
    -DBUILD_TESTS=OFF \
    -DBUILD_EXAMPLES=ON \
    ..
make -j4 && sudo make install
```
## Error behavior
At the make command, I received this error:
```
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/mnt/c/Users/source/stella_vslam/build'
make  -f src/stella_vslam/CMakeFiles/stella_vslam.dir/build.make src/stella_vslam/CMakeFiles/stella_vslam.dir/depend
make[2]: Entering directory '/mnt/c/Users/source/stella_vslam/build'
cd /mnt/c/Users/source/stella_vslam/build && /usr/local/lib/python3.10/dist-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /mnt/c/Users/source/stella_vslam /mnt/c/Users/source/stella_vslam/src/stella_vslam /mnt/c/Users/source/stella_vslam/build /mnt/c/Users/source/stella_vslam/build/src/stella_vslam /mnt/c/Users/source/stella_vslam/build/src/stella_vslam/CMakeFiles/stella_vslam.dir/DependInfo.cmake --color=                       Dependencies file "src/stella_vslam/CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o.d" is newer than depends file "/mnt/c/Users/source/stella_vslam/build/src/stella_vslam/CMakeFiles/stella_vslam.dir/compiler_depend.internal".
Consolidate compiler generated dependencies of target stella_vslam
make[2]: Leaving directory '/mnt/c/Users/source/stella_vslam/build'
make  -f src/stella_vslam/CMakeFiles/stella_vslam.dir/build.make src/stella_vslam/CMakeFiles/stella_vslam.dir/build
make[2]: Entering directory '/mnt/c/Users/source/stella_vslam/build'
[  1%] Building CXX object src/stella_vslam/CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o
cd /mnt/c/Users/source/stella_vslam/build/src/stella_vslam && /usr/bin/c++ -DFMT_LOCALE -DFMT_SHARED -DSPDLOG_ACTIVE_LEVEL=SPDLOG_LEVEL_DEBUG -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -DUSE_ARUCO -Dstella_vslam_EXPORTS -I/mnt/c/Users/source/stella_vslam/3rd/json/include -I/mnt/c/Users/source/stella_vslam/3rd/tinycolormap/include -I/mnt/c/Users/source/stella_vslam/3rd/spdlog/include -I/mnt/c/Users/source/stella_vslam/src -isystem /usr/include/suitesparse -isystem /usr/include/eigen3 -isystem /usr/include/opencv4 -isystem /usr/local/include/g2o/core -isystem /usr/local/include/g2o/stuff -isystem /usr/local/include/g2o/types/sba -isystem /usr/local/include/g2o/types/slam3d -isystem /usr/local/include/g2o/types/sim3 -isystem /usr/local/include/g2o/solvers/eigen -isystem /usr/local/include/g2o/solvers/csparse -O2 -g -DNDEBUG -std=gnu++17 -fPIC -Wall -Wextra -fopenmp -MD -MT src/stella_vslam/CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o -MF CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o.d -o CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o -c /mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/pose_optimizer_g2o.cc
In file included from /mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/perspective_pose_opt_edge.h:8,
from /mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/pose_opt_edge_wrapper.h:8,
from /mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/pose_optimizer_g2o.cc:6:
/mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/shot_vertex.h:28:26: error: ‘number_t’ does not name a type
28 |     void oplusImpl(const number_t* update_) override;                                                                                                                                                                                      |                          ^~~~~~~~
/mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/shot_vertex.h:28:10: error: ‘void stella_vslam::optimize::internal::se3::shot_vertex::oplusImpl(const int*)’ marked ‘override’, but does not override
28 |     void oplusImpl(const number_t* update_) override;                                                                                                                                                                                      |          ^~~~~~~~~
/mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/shot_vertex.h:57:42: error: ‘number_t’ does not name a type
57 | inline void shot_vertex::oplusImpl(const number_t* update_) {                                                                                                                                                                              |                                          ^~~~~~~~
/mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/shot_vertex.h: In member function ‘void stella_vslam::optimize::internal::se3::shot_vertex::oplusImpl(const int*)’:                                              /mnt/c/Users/source/stella_vslam/src/stella_vslam/optimize/internal/se3/shot_vertex.h:58:44: error: no matching function for call to ‘Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >::Map(const int*&)’
58 |     Eigen::Map<const Vec6_t> update(update_);                                                                                                                                                                                              |
```
There are other errors in the compilation, but most stem from this problem.
```
make[2]: *** [src/stella_vslam/CMakeFiles/stella_vslam.dir/build.make:804:
src/stella_vslam/CMakeFiles/stella_vslam.dir/optimize/pose_optimizer_g2o.cc.o] Error 1
make[2]: Leaving directory '/mnt/c/Users/source/stella_vslam/build'
make[1]: *** [CMakeFiles/Makefile2:460: src/stella_vslam/CMakeFiles/stella_vslam.dir/all] Error 2
make[1]: Leaving directory '/mnt/c/Users/source/stella_vslam/build'
make: *** [Makefile:136: all] Error 2
```

## Environment

- OS: Ubuntu 22.04
- OpenCV: 4.5.4
- Commit id: master
- Install procedure: https://stella-cv.readthedocs.io/en/latest/installation.html#build-instructions

ymd-stella, 2023-06-30:
This is caused by g2o version upgrade.

Nam-Nguyen-Hoang, 2023-06-30:
I did install g2o from the master branch: https://github.com/RainerKuemmerle/g2o, did it cause the problem?

ymd-stella, 2023-06-30:
I think it will work if you replace number_t with double.

Nam-Nguyen-Hoang, 2023-06-30:
Thank you very much, I will try it out. double or double_t?

ymd-stella, 2023-07-01:
Unless there is a specific reason not to, I recommend using g2o version 20230223_git.

ymd-stella, 2023-07-01:
Updated. You can now build using the latest version of g2o.

See https://github.com/stella-cv/stella_vslam/pull/514 for more information.

closed 2023-07-01 by ymd-stella