stella-cv/stella_vslam #532: CPU Usage using stella_vslam system

albertarla, 2023-09-14:
### Discussed in https://github.com/stella-cv/stella_vslam/discussions/530

<div type='discussions-op-text'>

<sup>Originally posted by **albertarla** September  5, 2023</sup>
Hi all,

I'm working in a project using stella_vslam with ros2. For that project I have a node that is publishing a camera_image topic, then I have a Component Manager that executes multiple components running stella_vslam (in multiple threads). The feed_image of stella_vslam is called inside the image callback.

If running 3 or 4 components the cpus usage is around the 60% of average and I'm receiving the images at 30fps as expected. After this point, if I start increasing the number of component being executed the fps received starts dropping. The thing is that the CPU is not at it's limit. 

I have done some tests to verify this is not an issue of ROS2 or it has something to do with the dds. Currently the publisher is using intra_process_comms with No copy of the image being done. This way I can ensure that the image published is getting to the subscribers as fast as posible. Moreover, running the ros2 topic hz i got 29fps so the issue is in the subscriber node.

#### Actual behaviour
As an example the results I'm getting are the following:

- 1 Component being run -> 30fps -> CPU 55%
- 2 Components being run -> 30fps -> CPU 55.5%
- ...
- 6 Components being run -> 20fps -> CPU 60%
- ...
- 8 Components being run -> 15fps -> CPU 70%
- ...
- 15 Components being run -> 4fps -> CPU 80%

#### Expected behaviour
I would expect that before decreasing the fps received in the components the CPU usage would increase, As an example the results I would expect are the following:

- 1 Component being run -> 30fps -> CPU 55%
- 2 Components being run -> 30fps -> CPU 55.5%
- ...
- 6 Components being run -> 30fps -> CPU 75%
- ...
- 8 Components being run -> 30fps -> CPU 90%
- ...
- 15 Components being run -> 10fps -> CPU 95%

#### Issue
So, the issue is that I think there is something in stella_vslam that limits the use of the CPU making the fps decrease before using all the capabilities available.

Any idea of what can be causing this?

</div>

albertarla, 2023-09-14:
Before opening this issue, i have investigated if it could be caused by ros2 or the dds in here: [https://github.com/ros2/rclcpp/issues/2295](url)

ymd-stella, 2023-09-14:
Insufficient explanation.

ymd-stella, 2023-09-14:
Close this issue because it does not follow the template.

closed 2023-09-14 by ymd-stella