stella-cv/stella_vslam #535: Multiple stella_vslam components deacrease performance

albertarla, 2023-10-11:
## Describe the bug

When executing multiple stella_vslam_ros components the computer does not use it's maximum capabilities. The performance (fps) starts decreassing when the cpu is far away from its limits and actually, it never gets close to its maximum capabilities. #532

 

## To Reproduce

 

 

To reproduce this follow the README.md provided in this repo: [https://github.com/movvo/rclcpp_max_consumption](https://github.com/movvo/rclcpp_max_consumption)

 

The tests done in the repo mentioned before can conclude that:
- ros2 components are working as expected as the CPU consumption is really similar as the one with single ros2 nodes. We have tested this by creating a component which uses a lot of CPU and then launching several of these components.
- executing single stella_vslam_ros nodes as separate processes doesn't deacrease de performance (fps) and the CPU increases as expected when multiple processes are launched. We have tested this by launching multiple stella vslam individual nodes at the same time.
- executing multiple stella_vslam_ros components in a single ros2 component container reduces the performance of the individual components (fpus). The CPU doesn't increase as expected and stays really low not using all the capacbilities of the CPU. We have tested this by launching the same stella vslam processes but as components of a component manager.

 

The components are multithreaded and some more experiments can be done in the repo. I tried with multiple number of components and using a ros2 component container isolated vs a multithread container. No significant diference found.

 

## Expected behavior

When launching multiple stella_vslam_ros components those will not deacrease they performance until the CPU of the computer reaches its maximum capabilities.

 

## Screenshots or videos

Can be found in the repo mentioned before ([url](https://github.com/movvo/rclcpp_max_consumption))

 

## Environment

 

- Hardware: PC Lenovo Legion 
- CPU: intel core i7-12700H 
- OS:Ubuntu 22.04
- Commit id: main branch of stella_vslam and main branch of stella_vslam_ros. The last one with a single modification for test porpoises.
- Install procedure: docker
- dataset: EuRoC MH_01

 

## Additional context

I have investigated if the problem could be because rclcpp_components but they seem to work as expected.

ymd-stella, 2023-10-11:
I don't see why you would want to launch multiple stella_vslam_ros processes that do exactly the same thing. Why is that?

albertarla, 2023-10-11:
The main reason is because mapping a large zone is quite difficult but creating multiple small maps is quick and have a better performance in relocalitzation tasks. 

Having multiple components, each one with a different map running is giving us better results than with a single large map. Moreover, if we know in which zone the system is, we can unload or load the components (unload or load a map) to reduce the total ram memory.

ymd-stella, 2023-10-11:
Multiple maps can be loaded into a process. See #380.

albertarla, 2023-10-13:
Here we have to separte things. On the one side, there is the issue that the current system tends to have scalability problems with large maps:
- When creating large maps: loop closing drops performance, takes a long time and have problems to correct the deviation, creating less accurate maps.
- When locating: relocalization takes a lot of time (probably because it has too many keyframes to check) and when it locates it has a lot of problems to keep the tracking and tends to get lost easily. 

We have tested these issues extensively in environments from 500m² to 40000m². According to our knowledge, this is something logical because of the increse of keyframes to check in the system.

 

The orignal issue of PR #380 was our first attempt to divide a large map into multiple small ones to adress the mapping issue and be able to map larger environments. Sorry for not finishing the PR and adding the sqlite version btw. 

 

To adress the second issue (when locating), the solution of separatnig `bow_database::keyfrms_in_node_` by map could improve the results as it would simulate what we are doing of launching multiple stella_vslam systems with a single map in each one. Anyway, we prefer the solution of having multiple stella_vslam system in separated components because this allows us to load and unload them whenever we need to.

On the other side, there is the problem of performance when using rclcpp components, which is the original topic of this issue and it is a different topic from the previous one. We are completely open to dicuss if there is a better way to target large maps, but at the moment our divide&conquer solution has given very good results and we are very happy with it. Basically, we divide the space into several maps, which are connected through a graph, each map is launched in a different component. When we are located in one of them, we keep the neighbour maps active, so relocalization is possible and we unload all the others. This works quite fine. However, when we don't know where we are (e.g. at the begining or when we are lost for more than certain ammount of time) we open all the small maps, here is where we see the dropping of performance.
It is clear that the performance of the component can be improved and that we are far away of being limited by the CPU, it makes no sense that the components drop their fps rate when we are still far away from the CPU limit. We are trying to understand the reason of this behaviour to try to address it. The repository I mention when creating the issue can reproduce the situation of having multiple stella_vslam components running and seeing that, even the CPU is not at its limit, the stella_vslam system is receiving the image at very low fps deacreasing the performance of the overall system.

ymd-stella, 2023-10-14:
The map loading/unloading should simply be implemented as a function of stella_vslam, without the use of components. Close this issue. If necessary, open another appropriate issue. For example

- Relocalization performance degrades on large maps
- Dynamic map loading/unloading

closed 2023-10-14 by ymd-stella