stella-cv/stella_vslam #543: Small bug with request_reset()

timbit123, 2023-11-09:
When we are trying to render a second video, we are creating a new system object and also just the be safe, we are also calling `request_reset()` but the reset is only called after the first frame is created (during `feed_frame`)

The issue is that `data::frame::next_id_` was not reset to 0 before de creation of the first frame in `create_monocular_frame`

we are manually resetting it, but I would suggest maybe adding something to reset the value during the system constructor?

Thank you
Tim

ymd-stella, 2023-11-10:
Essentially, the fact that data::frame::next_id_ is static seems to obscure the handling of initialization and destruction. i'll see if i can move it to system::next_id_. Thanks for the report.

ymd-stella, 2024-02-25:
Done. https://github.com/stella-cv/stella_vslam/pull/570

we can reset next_frame_id_ to 0 by destroying system.

In addition, `reset()` will not set next_frame_id_ to 0 after this change. I know this was an unnecessary and confusing feature, but if you need it, please let me know.

closed 2024-02-25 by ymd-stella