stella-cv/stella_vslam #578: Build fails on M1 Mac | Cannot find OpenMP

FreakTheMighty, 2024-04-05:
## Describe the bug
The build fails when attempting to build on an M1 mac when it fails to find OpenMP
`Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)`

## To Reproduce

* On an M1 mac running Mac os 14.4
* Install Mac dependencies according to these [instructions](https://stella-cv.readthedocs.io/en/latest/installation.html#installing-for-macos)
* Install common dependencies using these [instructions](https://stella-cv.readthedocs.io/en/latest/installation.html#common-installation-instructions)

Finally, when running
```shell-session
git clone --recursive https://github.com/stella-cv/stella_vslam.git
cd stella_vslam
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ..
make -j4
sudo make install
```

I get this output and error

```shell-session
make -j4
sudo make install

CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0074 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Build type: RelWithDebInfo
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization: DISABLED
CMake Error at /opt/homebrew/Cellar/cmake/3.29.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.29.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.29.1/share/cmake/Modules/FindOpenMP.cmake:581 (find_package_handle_standard_args)
  CMakeLists.txt:89 (find_package)
  ```

## Expected behavior
I'd like to get a successful build and install on MacOS

## Environment

- Hardware: M1 Macbook
- OS: MacOS 14.14
- Commit id: stella_vslam=b01b90022195bb15de3ffe2add0be4ce5229c065
- Install procedure: native

YYWumm, 2025-01-21:
I met the same problem but on intel Mac of macOs Sequoia, do you find a solution？

Nyanziba, 2025-02-21:
To add these command in CMakeLists.txt then I can build successfully.
```
if(CMAKE_C_COMPILER_ID MATCHES "Clang")
    set(OpenMP_C_FLAGS "-Xpreprocessor -fopenmp -I/opt/homebrew/opt/libomp/include")
    set(OpenMP_C_LIB_NAMES "omp")
    set(OpenMP_omp_LIBRARY "/opt/homebrew/opt/libomp/lib/libomp.dylib")
  endif()
  if(CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    set(OpenMP_CXX_FLAGS "-Xpreprocessor -fopenmp -I/opt/homebrew/opt/libomp/include")
    set(OpenMP_CXX_LIB_NAMES "omp")
    set(OpenMP_omp_LIBRARY "/opt/homebrew/opt/libomp/lib/libomp.dylib")
  endif()
```