stella-cv/stella_vslam #605: Missing aruco dependency during docker build

toddrjen, 2024-06-12:
## Describe the bug
The docker build is failing due to aruco not being installed with OpenCV on main

```shell-session
 => ERROR [18/20] RUN set -x &&   git clone https://github.com/stella-cv/stella_vslam_examples.git &&   cd stella_vslam_examples &&   git checkout -q b9d58a96ec5ae2434fd4ace5588c63984b0c1400 &&   git submodule update --init --recursiv  17.4s
------
 > [18/20] RUN set -x &&   git clone https://github.com/stella-cv/stella_vslam_examples.git &&   cd stella_vslam_examples &&   git checkout -q b9d58a96ec5ae2434fd4ace5588c63984b0c1400 &&   git submodule update --init --recursive &&   mkdir -p build &&   cd build &&   cmake     -DCMAKE_BUILD_TYPE=RelWithDebInfo     -DCMAKE_INSTALL_PREFIX=/usr/local     -DUSE_STACK_TRACE_LOGGER=ON     .. &&   make -j1:
0.268 + git clone https://github.com/stella-cv/stella_vslam_examples.git
0.269 Cloning into 'stella_vslam_examples'...
0.635 + cd stella_vslam_examples
0.635 + git checkout -q b9d58a96ec5ae2434fd4ace5588c63984b0c1400
0.639 + git submodule update --init --recursive
0.665 Submodule '3rd/filesystem' (https://github.com/gulrak/filesystem.git) registered for path '3rd/filesystem'
0.670 Cloning into '/stella_vslam_examples/3rd/filesystem'...
1.297 Submodule path '3rd/filesystem': checked out 'cd6805e94dd5d6346be1b75a54cdc27787319dd2'
1.303 + mkdir -p build
1.304 + cd build
1.304 + cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/usr/local -DUSE_STACK_TRACE_LOGGER=ON ..
1.370 -- The CXX compiler identification is GNU 11.4.0
1.424 -- The C compiler identification is GNU 11.4.0
1.433 -- Detecting CXX compiler ABI info
1.516 -- Detecting CXX compiler ABI info - done
1.528 -- Check for working CXX compiler: /usr/bin/c++ - skipped
1.528 -- Detecting CXX compile features
1.529 -- Detecting CXX compile features - done
1.533 -- Detecting C compiler ABI info
1.606 -- Detecting C compiler ABI info - done
1.620 -- Check for working C compiler: /usr/bin/cc - skipped
1.620 -- Detecting C compile features
1.621 -- Detecting C compile features - done
1.624 -- Looking for pthread.h
1.711 -- Looking for pthread.h - found
1.712 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
1.790 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
1.792 -- Found Threads: TRUE  
2.052 -- Found OpenMP_C: -fopenmp (found version "4.5") 
2.150 -- Found OpenMP_CXX: -fopenmp (found version "4.5") 
2.151 -- Found OpenMP: TRUE (found version "4.5")  
2.155 -- Found OpenCV: /usr/local (found version "4.7.0") 
2.161 CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindOpenGL.cmake:315 (message):
2.161   Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
2.161   available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
2.161   cmake_policy command to set the policy and suppress this warning.
2.161 
2.161   FindOpenGL found both a legacy GL library:
2.161 
2.161     OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so
2.161 
2.161   and GLVND libraries for OpenGL and GLX:
2.161 
2.161     OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
2.161     OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so
2.161 
2.161   OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
2.161   compatibility with CMake 3.10 and below the legacy GL library will be used.
2.161 Call Stack (most recent call first):
2.161   /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
2.161   /usr/local/lib/cmake/g2o/g2oConfig.cmake:4 (find_dependency)
2.161   /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
2.161   /usr/local/lib/cmake/stella_vslam/stella_vslamConfig.cmake:7 (find_dependency)
2.161   CMakeLists.txt:6 (find_package)
2.161 This warning is for project developers.  Use -Wno-dev to suppress it.
2.161 
2.162 -- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
2.173 -- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR) 
2.174 -- Found libbfd: /usr/lib/x86_64-linux-gnu/libbfd.so  
2.176 -- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR) 
2.177 -- Found Backward: /usr/local/lib/backward  
2.177 -- Stack trace logger: ENABLED
2.177 -- Google Perftools: DISABLED
2.189 -- Found Iridescence: /usr/local/include/iridescence  
2.190 -- Viewer for examples: IridescenceViewer
2.191 -- Configuring done
2.229 -- Generating done
2.231 -- Build files have been written to: /stella_vslam_examples/build
2.237 + make -j1
2.277 [  5%] Building CXX object CMakeFiles/run_camera_slam.dir/src/run_camera_slam.cc.o
16.85 [ 11%] Linking CXX executable run_camera_slam
17.31 /usr/bin/ld: /usr/local/lib/libstella_vslam.so: undefined reference to `cv::aruco::RefineParameters::RefineParameters(float, float, bool)'
17.31 /usr/bin/ld: /usr/local/lib/libstella_vslam.so: undefined reference to `cv::aruco::getPredefinedDictionary(cv::aruco::PredefinedDictionaryType)'
17.31 /usr/bin/ld: /usr/local/lib/libstella_vslam.so: undefined reference to `cv::aruco::ArucoDetector::ArucoDetector(cv::aruco::Dictionary const&, cv::aruco::DetectorParameters const&, cv::aruco::RefineParameters const&)'
17.31 /usr/bin/ld: /usr/local/lib/libstella_vslam.so: undefined reference to `vtable for cv::aruco::ArucoDetector'
17.33 collect2: error: ld returned 1 exit status
17.33 make[2]: *** [CMakeFiles/run_camera_slam.dir/build.make:138: run_camera_slam] Error 1
17.33 make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/run_camera_slam.dir/all] Error 2
17.33 make: *** [Makefile:91: all] Error 2
```

## To Reproduce

The same error occurs with all three docker files

shell session:

```shell-session
docker build -t stella_vslam-iridescense -f Dockerfile.iridescense .
```

```shell-session
docker build -t stella_vslam-desktop -f Dockerfile.desktop .
```

```shell-session
docker build -t stella_vslam-socket -f Dockerfile.socket .
```

## Expected behavior
Docker build finishes

## Environment

- Hardware: PC
- CPU: Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
- OS: openSUSE Tumbleweed
- Commit id: stella_vslam=0a1f13e2b2ee3f63a8dc40601e6e5d59a4663111
- Install procedure: docker
- dataset: N/A

## Additional context
This issue begins with the following commit id: stella_vslam=74a03bb99999ed74a1bf2e47d38c6f8adefe25b1

It does not occur with earlier commits.

ymd-stella, 2024-06-12:
Thanks for the detailed report.

ymd-stella, 2024-06-12:
Fixed.

PR #606 (2024-06-12): Fix aruco for OpenCV>=4.7
Close #605
closed 2024-06-12 by ymd-stella